StanfordVL/iGibson #112: 3d2d standford dataset annotations

elhamAm, 2021-08-18:
When using the 3d2d standford dataset  in igibson, the segmentation of the categories is not possible. It would be maybe possible to add the categories manually to the mesh_up_z.obj but then even in the gibson_indoor_scene.py, no category function is implemented. Is this something that is being worked on? If not, do you have any guidance? Thank you in advance.

elhamAm, 2021-08-20:
Figured it out !

closed 2021-09-22 by ChengshuLi

eunsunlee, 2022-09-06:
@elhamAm Hi,  I was wondering if you could share how you implemented the category function!

elhamAm, 2022-09-06:
Hi I added you as a collaborator :) you should have an invite

Atellas23, 2023-02-20:
Hey @elhamAm, would it be possible for you to share how did you manage to run iGibson with the 3D2D dataset and render the semantic segmentation?

elhamAm, 2023-02-20:
Hey I did this a long time ago, so I don't know but I shared the repo I made so if the answer is there, it's there :)