StanfordVL/iGibson #114: Discrete pick/place actions

jadkins99, 2021-08-20:
Are there discrete actions in Gibson for tasks such as pick/place? The only discrete actions that I have found thus far are regarding navigation. I have been interested in the room rearrangement task but the only control I have found for the fetch robot is the 7-dimensional joint space. Are there other ways to control the arm of the fetch robot?

ChengshuLi, 2021-09-22:
We don't exactly have such discrete actions because everything is physically simulated. However, we do have a motion planning wrapper and example that you could use as a starting point. For example, you can command the end-effector to reach a 3D point in space and then close the gripper to grasp. 

https://github.com/StanfordVL/iGibson/blob/master/igibson/examples/demo/motion_planning_example.py
https://github.com/StanfordVL/iGibson/blob/master/igibson/utils/motion_planning_wrapper.py

mjlbach, 2021-11-12:
Duplicate of #78, we'll be providing motion primitives for fetch in the next major release

closed 2021-11-12 by mjlbach