StanfordVL/iGibson #115: Missing 'non_sampleable_categories.txt' ?

evanchugh, 2021-08-25:
Upon downloading the assets and running
`python -m igibson.examples.demo.demo_static`

I get the following error: FileNotFoundError: [Errno 2] No such file or directory: '/home/evan/miniconda3/envs/igibson/lib/python3.8/site-packages/igibson/data/ig_dataset/metadata/non_sampleable_categories.txt'

I've tried reinstalling to no avail.

ChengshuLi, 2021-09-02:
Could you try to uninstall cmake in your conda env?

fxia22, 2021-09-03:
can you try a standalone `from cmake import cmake` and in command line `cmake --version`?

DhruvaBansal00, 2021-09-15:
Not sure if this is helpful, but I was able to install iGibson from source and run the demo. The PyPI version still throws the `FileNotFoundError` mentioned earlier in this issue.

fxia22, 2021-09-15:
@DhruvaBansal00 thanks for reporting this, the pypi version has been updated.

ChengshuLi, 2021-09-22:
Please feel free to re-open the issue if the problem persists. Thanks!

closed 2021-09-22 by ChengshuLi