StanfordVL/iGibson #117: Fail to install igibson 2.0 on windows10

Reinhard-Tichy, 2021-08-29:
I have tried to install iGibson 2.0 under Windows10 environment with Anaconda, following the instruction in the [doc](http://svl.stanford.edu/igibson/docs/installation.html). However, everytime I use pip to install with `pip install igibson`, it will always fail at the step of `Building wheels for collected packages: igibson`.

Below is the error message if I have already installed cmake==3.21.2 with pip before in my virtual env:

```
Building wheels for collected packages: igibson
  Building wheel for igibson (PEP 517): started
  Building wheel for igibson (PEP 517): finished with status 'error'
  ERROR: Command errored out with exit status 1:
   command: 'D:\tianchangyao\Anaconda3\envs\igibson\python.exe' 'D:\tianchangyao\Anaconda3\envs\igibson\lib\site-packages\pip\_vendor\pep517\_in_process.py' build_wheel 'C:\Users\TIANCH~1\AppData\Local\Temp\tmpm7htzual'
       cwd: C:\Users\TIANCH~1\AppData\Local\Temp\pip-install-zrtsogss\igibson_ec2b98728a3045289d35686797206e04
  Complete output (3746 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build\lib.win-amd64-3.7
  creating build\lib.win-amd64-3.7\igibson
  copying igibson\simulator.py -> build\lib.win-amd64-3.7\igibson
  copying igibson\__init__.py -> build\lib.win-amd64-3.7\igibson
  creating build\lib.win-amd64-3.7\igibson\activity
  copying igibson\activity\activity_base.py -> build\lib.win-amd64-3.7\igibson\activity
  copying igibson\activity\bddl_backend.py -> build\lib.win-amd64-3.7\igibson\activity
  copying igibson\activity\get_obj_shelf_heights.py -> build\lib.win-amd64-3.7\igibson\activity
  copying igibson\activity\__init__.py -> build\lib.win-amd64-3.7\igibson\activity
  creating build\lib.win-amd64-3.7\igibson\envs
  
  ..........................................(some irrelevant infomation has been omitted)

  running build_ext
  Traceback (most recent call last):
    File "d:\tianchangyao\anaconda3\envs\igibson\lib\runpy.py", line 193, in _run_module_as_main
      "__main__", mod_spec)
    File "d:\tianchangyao\anaconda3\envs\igibson\lib\runpy.py", line 85, in _run_code
      exec(code, run_globals)
    File "D:\tianchangyao\Anaconda3\envs\igibson\Scripts\cmake.exe\__main__.py", line 4, in <module>
  ModuleNotFoundError: No module named 'cmake'
  Traceback (most recent call last):
    File "D:\tianchangyao\Anaconda3\envs\igibson\lib\site-packages\pip\_vendor\pep517\_in_process.py", line 280, in <module>
      main()
    File "D:\tianchangyao\Anaconda3\envs\igibson\lib\site-packages\pip\_vendor\pep517\_in_process.py", line 263, in main
      json_out['return_val'] = hook(**hook_input['kwargs'])
    File "D:\tianchangyao\Anaconda3\envs\igibson\lib\site-packages\pip\_vendor\pep517\_in_process.py", line 205, in build_wheel
      metadata_directory)
    File "C:\Users\TIANCH~1\AppData\Local\Temp\pip-build-env-xtos8tyi\overlay\Lib\site-packages\setuptools\build_meta.py", line 222, in build_wheel
      wheel_directory, config_settings)
    File "C:\Users\TIANCH~1\AppData\Local\Temp\pip-build-env-xtos8tyi\overlay\Lib\site-packages\setuptools\build_meta.py", line 207, in _build_with_temp_dir
      self.run_setup()
    File "C:\Users\TIANCH~1\AppData\Local\Temp\pip-build-env-xtos8tyi\overlay\Lib\site-packages\setuptools\build_meta.py", line 259, in run_setup
      self).run_setup(setup_script=setup_script)
    File "C:\Users\TIANCH~1\AppData\Local\Temp\pip-build-env-xtos8tyi\overlay\Lib\site-packages\setuptools\build_meta.py", line 150, in run_setup
      exec(compile(code, __file__, 'exec'), locals())
    File "setup.py", line 169, in <module>
      include_package_data=True,
    File "C:\Users\TIANCH~1\AppData\Local\Temp\pip-build-env-xtos8tyi\overlay\Lib\site-packages\setuptools\__init__.py", line 153, in setup
      return distutils.core.setup(**attrs)
    File "D:\tianchangyao\Anaconda3\envs\igibson\lib\distutils\core.py", line 148, in setup
      dist.run_commands()
    File "D:\tianchangyao\Anaconda3\envs\igibson\lib\distutils\dist.py", line 966, in run_commands
      self.run_command(cmd)
    File "D:\tianchangyao\Anaconda3\envs\igibson\lib\distutils\dist.py", line 985, in run_command
      cmd_obj.run()
    File "C:\Users\TIANCH~1\AppData\Local\Temp\pip-build-env-xtos8tyi\overlay\Lib\site-packages\wheel\bdist_wheel.py", line 299, in run
      self.run_command('build')
    File "D:\tianchangyao\Anaconda3\envs\igibson\lib\distutils\cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "D:\tianchangyao\Anaconda3\envs\igibson\lib\distutils\dist.py", line 985, in run_command
      cmd_obj.run()
    File "D:\tianchangyao\Anaconda3\envs\igibson\lib\distutils\command\build.py", line 135, in run
      self.run_command(cmd_name)
    File "D:\tianchangyao\Anaconda3\envs\igibson\lib\distutils\cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "D:\tianchangyao\Anaconda3\envs\igibson\lib\distutils\dist.py", line 985, in run_command
      cmd_obj.run()
    File "setup.py", line 44, in run
      out = subprocess.check_output(["cmake", "--version"])
    File "D:\tianchangyao\Anaconda3\envs\igibson\lib\subprocess.py", line 411, in check_output
      **kwargs).stdout
    File "D:\tianchangyao\Anaconda3\envs\igibson\lib\subprocess.py", line 512, in run
      output=stdout, stderr=stderr)
  subprocess.CalledProcessError: Command '['cmake', '--version']' returned non-zero exit status 1.
  ----------------------------------------
  ERROR: Failed building wheel for igibson
Failed to build igibson
ERROR: Could not build wheels for igibson which use PEP 517 and cannot be installed directly
```

So could anyone help explain what has happened here?? Is the version of `cmake` not compatible with iGibson? I'm quite confused.

cgokmen, 2021-08-29:
Can you try again after uninstalling the version of cmake you have and installing the latest conda version with `conda install cmake`?

Reinhard-Tichy, 2021-08-30:
Still failed... However, the error message is different, it seems that `'pip` could not find where the cmake is?

Below is the error message:

```
Collecting igibson
  Using cached igibson-2.0.0.tar.gz (52.8 MB)
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
    Preparing wheel metadata: started
    Preparing wheel metadata: finished with status 'done'
Collecting gym>=0.12
  Using cached gym-0.19.0-py3-none-any.whl
Collecting freetype-py
  Using cached freetype_py-2.2.0-py3-none-win_amd64.whl (708 kB)
Collecting sphinx-markdown-tables
  Using cached sphinx_markdown_tables-0.0.15-py3-none-any.whl (28 kB)
Collecting py360convert
  Using cached py360convert-0.1.0-py3-none-any.whl
Collecting trimesh
  Using cached trimesh-3.9.29-py3-none-any.whl (635 kB)
Collecting gitpython
  Using cached GitPython-3.1.18-py3-none-any.whl (170 kB)
Collecting scipy>=1.2.1
  Using cached scipy-1.7.1-cp37-cp37m-win_amd64.whl (33.6 MB)
Collecting opencv-python>=3.4.8
  Using cached opencv_python-4.5.3.56-cp37-cp37m-win_amd64.whl (34.9 MB)
Collecting future
  Using cached future-0.18.2-py3-none-any.whl
Collecting PyYAML
  Using cached PyYAML-5.4.1-cp37-cp37m-win_amd64.whl (210 kB)
Collecting sphinx-rtd-theme
  Using cached sphinx_rtd_theme-0.5.2-py2.py3-none-any.whl (9.1 MB)
Collecting aenum
  Using cached aenum-3.1.0-py3-none-any.whl (123 kB)
Collecting tqdm
  Using cached tqdm-4.62.2-py2.py3-none-any.whl (76 kB)
Collecting transforms3d>=0.3.1
  Using cached transforms3d-0.3.1-py3-none-any.whl
Collecting ipython
  Using cached ipython-7.27.0-py3-none-any.whl (787 kB)
Collecting matplotlib
  Using cached matplotlib-3.4.3-cp37-cp37m-win_amd64.whl (7.2 MB)
Collecting GPUtil
  Using cached GPUtil-1.4.0-py3-none-any.whl
Collecting Pillow>=5.4.0
  Using cached Pillow-8.3.1-1-cp37-cp37m-win_amd64.whl (3.2 MB)
Collecting recommonmark
  Using cached recommonmark-0.7.1-py2.py3-none-any.whl (10 kB)
Collecting cloudpickle
  Using cached cloudpickle-1.6.0-py3-none-any.whl (23 kB)
Collecting networkx>=2.0
  Using cached networkx-2.6.2-py3-none-any.whl (1.9 MB)
Collecting numpy>=1.16.0
  Using cached numpy-1.21.2-cp37-cp37m-win_amd64.whl (14.0 MB)
Collecting h5py
  Using cached h5py-3.4.0-cp37-cp37m-win_amd64.whl (2.7 MB)
Collecting bddl
  Using cached bddl-0.0.3-py3-none-any.whl
Collecting sphinx>=1.8.0
  Using cached Sphinx-4.1.2-py3-none-any.whl (2.9 MB)
Collecting pybullet-svl>=3.1.6
  Using cached pybullet_svl-3.1.6.1-cp37-cp37m-win_amd64.whl
Collecting pytest
  Using cached pytest-6.2.4-py3-none-any.whl (280 kB)
Collecting sphinxcontrib-serializinghtml>=1.1.5
  Using cached sphinxcontrib_serializinghtml-1.1.5-py2.py3-none-any.whl (94 kB)
Collecting imagesize
[... 502 lines cut ...]
    validat6.cpp
    validat7.cpp
    validat8.cpp
    validat9.cpp
    validat10.cpp
    regtest1.cpp
    regtest2.cpp
    regtest3.cpp
    regtest4.cpp
    datatest.cpp
    正在生成代码...
    正在编译...
    fipsalgt.cpp
    fipstest.cpp
    dlltest.cpp
    正在生成代码...
    cryptest.vcxproj -> C:\Users\tianchangyao\AppData\Local\Temp\pip-install-pfcni0o6\igibson_a80771ce5e854f66bdfaae145d87c055\build\lib.win-amd64-3.7\igibson\render\mesh_renderer\build\Release\cryptest.exe
    Building Custom Rule C:/Users/tianchangyao/AppData/Local/Temp/pip-install-pfcni0o6/igibson_a80771ce5e854f66bdfaae145d87c055/igibson/render/CMakeLists.txt
  error: [WinError 3] 系统找不到指定的路径。: 'C:\\Users\\TIANCH~1\\AppData\\Local\\Temp\\pip-install-pfcni0o6\\igibson_a80771ce5e854f66bdfaae145d87c055\\igibson\\render\\mesh_renderer\\Release'
  ----------------------------------------
  ERROR: Failed building wheel for igibson
Failed to build igibson
ERROR: Could not build wheels for igibson which use PEP 517 and cannot be installed directly

```

Reinhard-Tichy, 2021-08-30:
Problem solved using the [3rd method](http://svl.stanford.edu/igibson/docs/installation.html#compile-from-source) instead of the 1st one.

closed 2021-08-30 by Reinhard-Tichy