StanfordVL/iGibson #118: installing on mac

Alpslee, 2021-09-01:
I use pip install igibson,then when I run the demo,I get this:
python -m igibson.examples.demo.demo_static

Traceback (most recent call last):
  File "/opt/anaconda3/envs/igibson/lib/python3.8/runpy.py", line 185, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/opt/anaconda3/envs/igibson/lib/python3.8/runpy.py", line 111, in _get_module_details
    __import__(pkg_name)
  File "/opt/anaconda3/envs/igibson/lib/python3.8/site-packages/igibson/__init__.py", line 11, in <module>
    global_config = yaml.load(f, Loader=yaml.FullLoader)
  File "/opt/anaconda3/envs/igibson/lib/python3.8/site-packages/yaml/__init__.py", line 114, in load
    return loader.get_single_data()
File "/opt/anaconda3/envs/igibson/lib/python3.8/site-packages/yaml/constructor.py", line 49, in get_single_data
    node = self.get_single_node()
  File "/opt/anaconda3/envs/igibson/lib/python3.8/site-packages/yaml/composer.py", line 39, in get_single_node
    if not self.check_event(StreamEndEvent):
  File "/opt/anaconda3/envs/igibson/lib/python3.8/site-packages/yaml/parser.py", line 98, in check_event
    self.current_event = self.state()
  File "/opt/anaconda3/envs/igibson/lib/python3.8/site-packages/yaml/parser.py", line 171, in parse_document_start
    raise ParserError(None, None,
yaml.parser.ParserError: expected '<document start>', but found '<scalar>'
  in "/opt/anaconda3/envs/igibson/lib/python3.8/site-packages/igibson/global_config.yaml", line 2, column 1
(igibson) wuweilindeMacBook-Pro:~ Rand$ python -m igibson.examples.demo.demo_static
Traceback (most recent call last):
  File "/opt/anaconda3/envs/igibson/lib/python3.8/runpy.py", line 185, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/opt/anaconda3/envs/igibson/lib/python3.8/runpy.py", line 111, in _get_module_details
    __import__(pkg_name)
  File "/opt/anaconda3/envs/igibson/lib/python3.8/site-packages/igibson/__init__.py", line 11, in <module>
    global_config = yaml.load(f, Loader=yaml.FullLoader)
  File "/opt/anaconda3/envs/igibson/lib/python3.8/site-packages/yaml/__init__.py", line 114, in load
    return loader.get_single_data()
  File "/opt/anaconda3/envs/igibson/lib/python3.8/site-packages/yaml/constructor.py", line 49, in get_single_data
    node = self.get_single_node()
  File "/opt/anaconda3/envs/igibson/lib/python3.8/site-packages/yaml/composer.py", line 39, in get_single_node
    if not self.check_event(StreamEndEvent):
  File "/opt/anaconda3/envs/igibson/lib/python3.8/site-packages/yaml/parser.py", line 98, in check_event
    self.current_event = self.state()
  File "/opt/anaconda3/envs/igibson/lib/python3.8/site-packages/yaml/parser.py", line 171, in parse_document_start
    raise ParserError(None, None,
yaml.parser.ParserError: expected '<document start>', but found '<scalar>'
  in "/opt/anaconda3/envs/igibson/lib/python3.8/site-packages/igibson/global_config.yaml", line 2, column 1
 
is there something wrong with the igibson/global_config.yaml  file? please help me.

ChengshuLi, 2021-09-01:
I am not sure I can reproduce this issue.

Could you try the following? There might be a YAML version issue.
```
import yaml
print(yaml.__version__)
with open('/opt/anaconda3/envs/igibson/lib/python3.8/site-packages/igibson/global_config.yaml') as f:
    yaml.load(f, Loader=yaml.FullLoader)
```

Alpslee, 2021-09-01:
> I am not sure I can reproduce this issue.
> 
> Could you try the following? There might be a YAML version issue.
> 
> ```
> import yaml
> print(yaml.__version__)
> with open('/opt/anaconda3/envs/igibson/lib/python3.8/site-packages/igibson/global_config.yaml') as f:
>     yaml.load(f, Loader=yaml.FullLoader)
> ```

Thanks.I followed it then I got this:

import yaml
>>> print(yaml.__version__)
5.4.1
>>> with open('/opt/anaconda3/envs/igibson/lib/python3.8/site-packages/igibson/global_config.yaml') as f:
...     yaml.load(f, Loader=yaml.FullLoader)
... 
Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
  File "/opt/anaconda3/envs/igibson/lib/python3.8/site-packages/yaml/__init__.py", line 114, in load
    return loader.get_single_data()
  File "/opt/anaconda3/envs/igibson/lib/python3.8/site-packages/yaml/constructor.py", line 49, in get_single_data
    node = self.get_single_node()
  File "/opt/anaconda3/envs/igibson/lib/python3.8/site-packages/yaml/composer.py", line 39, in get_single_node
    if not self.check_event(StreamEndEvent):
  File "/opt/anaconda3/envs/igibson/lib/python3.8/site-packages/yaml/parser.py", line 98, in check_event
    self.current_event = self.state()
  File "/opt/anaconda3/envs/igibson/lib/python3.8/site-packages/yaml/parser.py", line 171, in parse_document_start
    raise ParserError(None, None,
yaml.parser.ParserError: expected '<document start>', but found '<scalar>'
  in "/opt/anaconda3/envs/igibson/lib/python3.8/site-packages/igibson/global_config.yaml", line 2, column 1

Bty,I edited my global_config.yaml file with Xcode ,it looks like that:

assets_path:/opt/anaconda3/envs/igibson/lib/python3.8/site-packages/igibson/data/assets                            # put either absolute path or relative to current directory
g_dataset_path:/opt/anaconda3/envs/igibson/lib/python3.8/site-packages/igibson/data/g_dataset
ig_dataset_path: /opt/anaconda3/envs/igibson/lib/python3.8/site-packages/igibson/data/ig_dataset
threedfront_dataset_path: /opt/anaconda3/envs/igibson/lib/python3.8/site-packages/igibson/data/threedfront_dataset
cubicasa_dataset_path: /opt/anaconda3/envs/igibson/lib/python3.8/site-packages/igibson/data/cubicasa_dataset
key_path: data/igibson.key

fxia22, 2021-09-01:
Thanks for reporting the issue, looks like you yaml has wrong syntax. It should be like this (be careful with the space after :)
```yaml
assets_path: /opt/anaconda3/envs/igibson/lib/python3.8/site-packages/igibson/data/assets # put either absolute path or relative to current directory
g_dataset_path: /opt/anaconda3/envs/igibson/lib/python3.8/site-packages/igibson/data/g_dataset
ig_dataset_path: /opt/anaconda3/envs/igibson/lib/python3.8/site-packages/igibson/data/ig_dataset
threedfront_dataset_path: /opt/anaconda3/envs/igibson/lib/python3.8/site-packages/igibson/data/threedfront_dataset
cubicasa_dataset_path: /opt/anaconda3/envs/igibson/lib/python3.8/site-packages/igibson/data/cubicasa_dataset
key_path: data/igibson.key

```

Alpslee, 2021-09-02:
> Thanks for reporting the issue, looks like you yaml has wrong syntax. It should be like this (be careful with the space after :)
>
> ```yaml
> assets_path: [...]
thanks.A little bit different result for this time:

with open(os.path.join(igibson.ig_dataset_path, "metadata/non_sampleable_categories.txt")) as f:
FileNotFoundError: [Errno 2] No such file or directory: '/opt/anaconda3/envs/igibson/lib/python3.8/site-packages/igibson/data/ig_dataset/metadata/non_sampleable_categories.txt'

seems like I need to download the dataset first?

ChengshuLi, 2021-09-02:
Related to #115 
@fxia22 Would you mind pushing the latest `master` to pypi? Thanks!

closed 2021-09-04 by Alpslee

DhruvaBansal00, 2021-09-15:
I am having the same issue on Ubuntu 20.04. Has the latest `master` been pushed to pypi?

fxia22, 2021-09-15:
@ChengshuLi done