StanfordVL/iGibson #119: How to decode BEHAVIOR Dataset ?

ldy8665, 2021-09-03:
I use the **docker image** of igibon2.
I get the **igibson.key** and put it in **/opt/igibson/gibson2/data**  as instructions in http://svl.stanford.edu/igibson/docs/dataset.html.
But when i ran the demo of PBR (Physics-Based Rendering) Example as instructions (http://svl.stanford.edu/igibson/docs/renderer.html), i got the error with 
"UnicodeDecodeError: 'ascii' codec can't decode byte 0x97 in position 0: ordinal not in range(128)
"
So i think .obj file is not been decoded.
So how to decode BEHAVIOR Dataset when i use it ?

fxia22, 2021-09-03:
It could be the docker image is not up-to-date with the support of encrypted assets. Can you try updating the iGibson to the head of master branch in the docker image? In the mean time, we will update the docker image.

ldy8665, 2021-09-03:
Thank u bro! 
And also files in github: **iGibson/docker/pull.images.s**h need to been updated.
I`ll try to use iGibson2.0 installed by pip first.
Waiting for ur new docker image.
Thank u again ！

ldy8665, 2021-09-03:
> It could be the docker image is not up-to-date with the support of encrypted assets. Can you try updating the iGibson to the head of master branch in the docker image? In the mean time, we will update the docker image.

Thank u bro!
And also files in github: iGibson/docker/pull.images.sh need to been updated.
I`ll try to use iGibson2.0 installed by pip first.
Waiting for ur new docker image.
Thank u again ！

mjlbach, 2021-11-13:
I've updated the containers from the latest master. In the next release, all containers will be automatically built with a github action so this shouldn't happen again.

closed 2021-11-13 by mjlbach