StanfordVL/iGibson #120: Installation Issue

Qiwei97, 2021-09-03:
Hi, I am trying to install igibson on a virtualenv in a remote jupyter server using ! pip install git+https://github.com/StanfordVL/iGibson.git but I keep getting the below error. I have attempted to upgrade setuptools and reinstalling cmake by doing pip install cmake but it doesnt seem to rectify the problem.

Traceback (most recent call last):
    File "/home2/linqiwei/pythonprojects/python_env/bin/cmake", line 5, in <module>
      from cmake import cmake
  ModuleNotFoundError: No module named 'cmake'
  Traceback (most recent call last):
    File "/home2/linqiwei/pythonprojects/python_env/lib/python3.6/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 349, in <module>
      main()
    File "/home2/linqiwei/pythonprojects/python_env/lib/python3.6/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 331, in main
      json_out['return_val'] = hook(**hook_input['kwargs'])
    File "/home2/linqiwei/pythonprojects/python_env/lib/python3.6/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 249, in build_wheel
      metadata_directory)
    File "/tmp/pip-build-env-moqcx67k/overlay/lib/python3.6/site-packages/setuptools/build_meta.py", line 222, in build_wheel
      wheel_directory, config_settings)
    File "/tmp/pip-build-env-moqcx67k/overlay/lib/python3.6/site-packages/setuptools/build_meta.py", line 207, in _build_with_temp_dir
      self.run_setup()
    File "/tmp/pip-build-env-moqcx67k/overlay/lib/python3.6/site-packages/setuptools/build_meta.py", line 259, in run_setup
      self).run_setup(setup_script=setup_script)
    File "/tmp/pip-build-env-moqcx67k/overlay/lib/python3.6/site-packages/setuptools/build_meta.py", line 150, in run_setup
      exec(compile(code, __file__, 'exec'), locals())
    File "setup.py", line 169, in <module>
      include_package_data=True,
    File "/tmp/pip-build-env-moqcx67k/overlay/lib/python3.6/site-packages/setuptools/__init__.py", line 153, in setup
      return distutils.core.setup(**attrs)
    File "/usr/lib/python3.6/distutils/core.py", line 148, in setup
      dist.run_commands()
    File "/usr/lib/python3.6/distutils/dist.py", line 955, in run_commands
      self.run_command(cmd)
    File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
      cmd_obj.run()
    File "/tmp/pip-build-env-moqcx67k/overlay/lib/python3.6/site-packages/wheel/bdist_wheel.py", line 299, in run
      self.run_command('build')
    File "/usr/lib/python3.6/distutils/cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
      cmd_obj.run()
    File "/usr/lib/python3.6/distutils/command/build.py", line 135, in run
      self.run_command(cmd_name)
    File "/usr/lib/python3.6/distutils/cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
      cmd_obj.run()
    File "setup.py", line 44, in run
      out = subprocess.check_output(["cmake", "--version"])
    File "/usr/lib/python3.6/subprocess.py", line 356, in check_output
      **kwargs).stdout
    File "/usr/lib/python3.6/subprocess.py", line 438, in run
      output=stdout, stderr=stderr)
  subprocess.CalledProcessError: Command '['cmake', '--version']' returned non-zero exit status 1.
  ----------------------------------------
  ERROR: Failed building wheel for igibson
Failed to build igibson
ERROR: Could not build wheels for igibson which use PEP 517 and cannot be installed directly

fxia22, 2021-09-03:
It is related to cmake. If `pip install cmake` doesn't work, can you install cmake with `apt`? For example `sudo apt install cmake`.

Qiwei97, 2021-09-03:
Hi, thanks for the prompt reply. I managed to get it working by using conda install instead. However, I do have another issue that popped up. 

Upon running `! python -m igibson.utils.assets_utils --download_assets`, I am getting the below error:

> INFO:root:Importing iGibson (igibson module)
INFO:root:Assets path: C:\Users\linqw\iGibson\igibson\data/assets
INFO:root:Gibson Dataset path: C:\Users\linqw\iGibson\igibson\data/g_dataset
INFO:root:iG Dataset path: C:\Users\linqw\iGibson\igibson\data/ig_dataset
INFO:root:3D-FRONT Dataset path: C:\Users\linqw\iGibson\igibson\data/threedfront_dataset
INFO:root:CubiCasa5K Dataset path: C:\Users\linqw\iGibson\igibson\data/cubicasa_dataset
INFO:root:iGibson Key path: C:\Users\linqw\iGibson\igibson\data/igibson.key
INFO:root:Example path: C:\Users\linqw\iGibson\igibson\examples
INFO:root:Example config path: C:\Users\linqw\iGibson\igibson\examples\configs
/tmp/assets_igibson.tar.gz: No such file or directory
tar: Error opening archive: Failed to open '/tmp/assets_igibson.tar.gz'

May I know what can be done to rectify this?

fxia22, 2021-09-03:
I see, looks like you are running on windows, you can directly download https://storage.googleapis.com/gibson_scenes/assets_igibson.tar.gz  and place it in `<your path>\iGibson\igibson\data/assets`

Qiwei97, 2021-09-04:
Hi, Thanks! It is working. However, when I tried running the code `!python -m igibson.examples.demo.env_example`, I am getting the below error.

> INFO: Succesfully initialized GLFW context and window!
Compiling GLSL shader: c:\users\linqw\igibson\igibson\render\mesh_renderer\shaders\450\equirect2cube_cs.glsl
Compiling GLSL shader: c:\users\linqw\igibson\igibson\render\mesh_renderer\shaders\450\spmap_cs.glsl
Compiling GLSL shader: c:\users\linqw\igibson\igibson\render\mesh_renderer\shaders\450\irmap_cs.glsl
Compiling GLSL shader: c:\users\linqw\igibson\igibson\render\mesh_renderer\shaders\450\spbrdf_cs.glsl
Loading image: C:\Users\linqw\iGibson\igibson\data/ig_dataset\scenes\background\photo_studio_01_2k.hdr
INFO: compiled pbr shaders
starting thread 0
started testThreads thread 0 with threadHandle 0000000000000C14
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
Version = 4.6.0 NVIDIA 462.80
Vendor = NVIDIA Corporation
Renderer = GeForce MX250/PCIe/SSE2
b3Printf: Selected demo: Physics Server
starting thread 0
started MotionThreads thread 0 with threadHandle 0000000000000CC8
MotionThreadFunc thread started
numActiveThreads = 0
stopping threads
Thread with taskId 0 with handle 0000000000000CC8 exiting
Thread TERMINATED
finished
show more (open the raw output data in a text editor) ...

finished
numActiveThreads = 0
btShutDownExampleBrowser stopping threads
Thread with taskId 0 with handle 0000000000000C14 exiting
Thread TERMINATED
INFO:root:Importing iGibson (igibson module)
INFO:root:Assets path: C:\Users\linqw\iGibson\igibson\data/assets
INFO:root:Gibson Dataset path: C:\Users\linqw\iGibson\igibson\data/g_dataset
INFO:root:iG Dataset path: C:\Users\linqw\iGibson\igibson\data/ig_dataset
INFO:root:3D-FRONT Dataset path: C:\Users\linqw\iGibson\igibson\data/threedfront_dataset
INFO:root:CubiCasa5K Dataset path: C:\Users\linqw\iGibson\igibson\data/cubicasa_dataset
INFO:root:iGibson Key path: C:\Users\linqw\iGibson\igibson\data/igibson.key
INFO:root:Example path: C:\Users\linqw\iGibson\igibson\examples
INFO:root:Example config path: C:\Users\linqw\iGibson\igibson\examples\configs
pybullet build time: Sep  3 2021 00:17:52
INFO:root:Loading C:\Users\linqw\iGibson\igibson\data/assets\models/mjcf_primitives/cube.obj
INFO:root:Loading C:\Users\linqw\iGibson\igibson\data/assets\models/mjcf_primitives/sphere8.obj
Traceback (most recent call last):
  File "C:\Users\linqw\anaconda3\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\linqw\anaconda3\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "c:\users\linqw\igibson\igibson\examples\demo\env_example.py", line 25, in <module>
    main()
  File "c:\users\linqw\igibson\igibson\examples\demo\env_example.py", line 11, in main
    env = iGibsonEnv(config_file=config_filename, mode="gui")
  File "c:\users\linqw\igibson\igibson\envs\igibson_env.py", line 53, in __init__
    super(iGibsonEnv, self).__init__(
  File "c:\users\linqw\igibson\igibson\envs\env_base.py", line 67, in __init__
    self.simulator = Simulator(
show more (open the raw output data in a text editor) ...

  File "c:\users\linqw\igibson\igibson\simulator.py", line 170, in __init__
    self.gen_assisted_grasping_categories()
  File "c:\users\linqw\igibson\igibson\simulator.py", line 1452, in gen_assisted_grasping_categories
    if v["enable_ag"]:
KeyError: 'enable_ag'

fxia22, 2021-09-04:
@Qiwei97 Looks like you are using iG 2.0 with the dataset for iG 1.0, you can download the iG 2.0 dataset following this guide: http://svl.stanford.edu/igibson/docs/dataset.html#download-igibson-2-0-scenes-and-the-behavior-dataset-of-objects

Qiwei97, 2021-09-06:
Alright thanks! Got it working!

closed 2021-09-06 by Qiwei97