StanfordVL/iGibson #122: ERROR:root:Optimized renderer is not supported on Mac

ArbiterGe, 2021-09-07:
In the Macbookair can't render in the enviroment when I Compile source code to install iGibson
when I run env_example.py
find problem：
WARNING:root:Rendering both iggui and pbgui is not supported on mac, choose either pbgui or iggui. Default to iggui.
INFO: Probing, EGL cannot run on this device
INFO:root:Using device 0 for rendering
ERROR:root:Optimized renderer is not supported on Mac

ArbiterGe, 2021-09-07:
My macbook is macbook air m1

shivanimall, 2021-09-20:
Facing the same issue as OP

ArbiterGe, 2021-09-20:
> Facing the same issue as OP when running any file that requires iggui eg: `mouse_interaction.py`, `env_interactive_example.py`
> noting that have been successfully connecting to pybullet GUI.
I set the parameter “optimization” to false

ArbiterGe, 2021-09-20:
I set the parameter "optimization" to false

shivanimall, 2021-09-20:
which file is that?

ChengshuLi, 2021-09-21:
Feel free to try this. 
```
settings = MeshRendererSettings(optimized=False)
simulator = Simulator(rendering_settings=settings)
```

closed 2021-09-21 by ChengshuLi