StanfordVL/iGibson #123: Parallel Execution of plans for multiple robots

krsandeep98, 2021-09-08:
Hi @fxia22 @ChengshuLi ,

Thanks a lot for putting in so much effort in creating this awesome simulator.

I am trying to load multiple robots and do motion planning for them and visualize the result in the GUI. With some help from the **plan_base_motion** and **dry_run_base_plan** functions of the **motion_planning_wrapper.py** file in the **igibson/utils** folder, I have been able to set up a simple point navigation setup in the gibson Rs room environment. 

However, **I want to execute the plans for the robots simultaneously and not sequentially** (calling dry_run_base_plan for each of them one after the other). I was looking at the multiprocessing library in python and tried using multiple Processes with different functions corresponding to different robots. 

But I am not sure why there is **no change in the environment state when I am calling the dry_run_base_plan functions with the help of the multiprocessing library**. 

I looked at some of the issues related to multiprocessing library in the repository, but they are concerned with having multiple environments for the different processes. I want to just run multiple functions in the same environment and any help on this would be greatly appreciated.

This is the outline of what I am trying to do

```
#functions for executing the paths
dry_run_base_robot1(env, path)
    for waypoint in path:
         set_base_values(robot1, waypoint)

dry_run_base_robot2(env, path)
     ...

#main function
env = iGibsonEnv(config_file, mode = 'gui')

path1 = plan_base_motion(robot1, goal)
path2 = plan_base_motion(robot2, goal)

processes = []
p1 = multiprocessing.Process(target=dry_run_base_robot1 , args=(env, path1))
processes.append(p1)
p1.start()

p2 = multiprocessing.Process(target= dry_run_base_robot2, args=(env, path2))
processes.append(p2)
p2.start()
        
for process in processes:
    process.join()
env.close()
```

This way of execution doesn't give me any outputs in the gui mode (the viewer and robot view just freeze and I have to kill the processes later). They run fine when using the headless mode, but the robot positions are not updated. Everything works perfectly if I just directly call the two dry_run_base functions one after the other without any multiprocessing (both in GUI and headless mode). 

I don't know if I am doing anything wrong, but please let me know if you have any ideas on how to fix this or approach it from a different perspective.

Thanks,
Sandeep

krsandeep98, 2021-09-09:
Update:

Taking inspiration from the `igibson_ray_env.py` file in the `envs` directory, I was able to get the different planning and execution functions added in the `igibson_env` file itself and now I can use `ray.get()` to call them remotely and it runs fine in both gui and headless mode. The positions of the robots are updated in both the cases.

But, all of this runs on a **single processor** and uses only one PID (the rest are IDLE in the ray dashboard). I tried experimenting with different values of the `num_cpus` argument, but it's still the same. Basically it's still running sequentially despite me using `ray.get(processes)`, I am not sure why that's the case. 

I tried using the same environment and packages to run 2 basic for loops (in a different example not involving the igibson simulator) and it runs well on multiple cores (as seen from the dashboard, all workers (2 for this example) are running and none of them is IDLE). 

Do you have any idea on how to make the code involving igibson simulator use **multiple processors**?

cc: @roberto-martinmartin 

Thanks,
Sandeep

ChengshuLi, 2021-09-22:
Hi Sandeep, have you tried using this? https://github.com/StanfordVL/iGibson/blob/master/igibson/envs/parallel_env.py

fxia22, 2021-09-22:
@krsandeep98 I am wondering why do you need to use multiple process in this case? I think multiple robot can operate in the same process, something like below:
```
for ...:
    robot1.apply_action(...)
    robot2.apply_action(...)
    simulator.step()
```

If you are using the motion planner, you can still execute the motions for each robot. Does this work for you?

krsandeep98, 2021-09-22:
Hi Chengshu and Fei,

Thanks a lot for getting back on this.

I did come across this earlier and thought that this was not exactly what I wanted since I want one environment and multiple robots to take actions in there. Also the comments in that file suggest that all the environments should use the same action and observation spaces, so I was not sure if spinning up a separate environment for every robot was a good idea. Please let me know if I am missing something and you had a different way in which the parallel environments could be used.

Fei,
This is exactly what I am doing as a workaround, basically combining the 2 dry run functions mentioned earlier in the issue. I had a few questions about that. 

1. Firstly, the `plan_base_motion` is giving me waypoints and I am not sure how to convert them to actions specific to the robot or add a time element to the waypoints based on the dynamics of the robot. Do you have any suggestions on how to go about that or some examples which I can use maybe?
2. I am trying to have a plan, execute, replan loop with the replanning part taking in the updated information from the environment. But since I am only able to run either the execution or the planning at a time (due to a single processor), when I am planning, the whole environment is static which is not realistic and it would be great to have the robots moving independently of the planning aspect of the robots. Does this make sense? Please let me know if I should elaborate on anything.

Thanks,
Sandeep