StanfordVL/iGibson #124: Issue with video display

ericrosenbrown, 2021-09-09:
Hello,

I have gone through the installation process and am trying to run the following command with the GUI:

python iGibson/igibson/examples/behavior/behavior_env_metrics.py

when I run it, I get a lot of print outs (objects are loaded, file sizes are reported, etc.) , but it generates a GUI window that just stays black the entire time. Should I be seeing something when I run this besides a black window, and if I don't see something, what would be causing this issue? 

If it helps, the last few lines printed before the program ends is:
Merged vertex data shape:
(3108540, 14)
Enable pbr: True
multidrawcount 481
Segmentation fault (core dumped)

mjlbach, 2021-09-09:
What platform/GPU are you on?

ericrosenbrown, 2021-09-09:
> What platform/GPU are you on?

My platform setup is:
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.6 LTS
Release:	16.04
Codename:	xenial

For my GPU setup:
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 384.130                Driver Version: 384.130                   |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  GeForce GTX 1070    Off  | 00000000:01:00.0  On |                  N/A |
|  0%   50C    P0    35W / 151W |    589MiB /  8112MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+

IfrahIdrees, 2021-09-15:
Hello, 

Is there a solution to this issue? I am facing exactly the same issue in headless mode.

mjlbach, 2021-09-15:
It's hard to debug your individual setups, I would first try to reproduce on an Ubuntu 20.04 workstation with up to date drivers:

This provides the dependencies on a fresh google cloud instance (which I've verified). Note we do test iGibson on Ubuntu 16/18/20.04 and Fedora 34/CentOS 8.

```bash
# Add the nvidia ubuntu repositories
apt-get update && apt-get install -y --no-install-recommends \
    gnupg2 curl ca-certificates && \
    curl -fsSL https://developer.download.nvidia.com/compute/cuda/repos/ubuntu2004/x86_64/7fa2af80.pub | apt-key add - && \
    echo "deb https://developer.download.nvidia.com/compute/cuda/repos/ubuntu2004/x86_64 /" > /etc/apt/sources.list.d/cuda.list && \
    echo "deb https://developer.download.nvidia.com/compute/machine-learning/repos/ubuntu2004/x86_64 /" > /etc/apt/sources.list.d/nvidia-ml.list

# The following cuda libraries are required to compile igibson
apt-get update && apt-get update && apt-get install -y --no-install-recommends \
    xserver-xorg-video-nvidia-470 \
    cuda-cudart-11-1=11.1.74-1 \
    cuda-compat-11-1 \
    cuda-command-line-tools-11-1=11.1.1-1 \
    cuda-libraries-dev-11-1=11.1.1-1 \

# For building and running igibson
apt-get update && apt-get install -y --no-install-recommends \
    cmake \
    git \
    g++ \
    libegl-dev
```

IfrahIdrees, 2021-09-15:
Actually, we both are on the same team and using the same lab computer. The only difference is I am sshing into the machine and require the headless mode. Our machine is ubuntu 16 and you said you tested on it so this problem so we should not have been faced this, right?

We do have nvidia and cuda libraries for cuda7.5 already installed. Will the version make a difference?

mjlbach, 2021-09-15:
Ah, that makes sense! I can try to find one of our lab computers running 16.04 later and verify.

For reference, this is what you should see (on the latest ig-develop)
[image]

I'd recommend upgrading your graphics drivers as they are extremely out of date, and we've run into various issues (most recently a cgroups issue affecting creating egl contexts on the 460.x and 465.x driver series)

mjlbach, 2021-11-12:
I've verified the latest iGibson work on ubuntu 16.04 with the correct dependencies from our [ installation guide](http://svl.stanford.edu/igibson/docs/installation.html). Please ensure your nvidia driver and libegl1 are up to date, there are reported issues with EGL context I mentioned before.

closed 2021-11-12 by mjlbach