StanfordVL/iGibson #125: EGL issue when sshing

ericrosenbrown, 2021-09-15:
We are participating in the BEHAVIOR challenge and have gone through the installation process for iGibson provided in the instructions. However, whenever we ssh into our machine with the setup and try to run a script, like:

python -m igibson.examples.demo.env_example

The script simply fails to run and gives the following error:

INFO: Probing, EGL cannot run on this device
INFO:root:Using device 0 for rendering
ERROR: Failed to EGL with glad.

We have received this issue when sshing into our machine using a Mac and a Ubuntu 16.04 machine. Is there something we need to do to make sure we can ssh into a computer and run iGibson correctly?

ChengshuLi, 2021-09-22:
https://github.com/StanfordVL/iGibson/blob/master/igibson/examples/demo/env_example.py uses "gui" mode, and I think you are accessing a headless server via ssh.

Try to use `mode='headless'` instead.

fxia22, 2021-09-22:
Hi Eric, it looks like your ubuntu machine doesn't have OpenGL/EGL installed?

You can take a look at our trouble shooting guide http://svl.stanford.edu/igibson/docs/issues.html
and installation guide: http://svl.stanford.edu/igibson/docs/installation.html

mjlbach, 2021-11-12:
Please reopen if EGL issues persist.

closed 2021-11-12 by mjlbach