StanfordVL/iGibson #127: problems when enabling optical_flow in iGibson 1.0

UltronAI, 2021-09-16:
Hi, I encountered several problems (possibly bugs) after I added optical_flow to sensors. 

I conducted my experiments with iGibson 1. 0 (commit: 893fcda1a1d01ce2da5102d500d2d16f35a62162)

First, I think there is a bug in https://github.com/StanfordVL/iGibson/blob/893fcda1a1d01ce2da5102d500d2d16f35a62162/gibson2/sensors/vision_sensor.py#L108 
 `return raw_vision_obs['optical_flow'][:, :, :3]`
should be 
`return raw_vision_obs['optical_flow'][:, :, :2]`

Second, I find if I want to use optical_flow, I have to change `optimized` to True in [MeshRendererSettings](https://github.com/StanfordVL/iGibson/blob/893fcda1a1d01ce2da5102d500d2d16f35a62162/gibson2/render/mesh_renderer/mesh_renderer_settings.py#L20), otherwise `self.trans_data` and `self.rot_data` in `MeshRenderer` will be None, leading to error for updating dynamic positions with need_flow_info=True. Then, vertex and texture will be optimized in the first time of loading scenes. Is it necessary to do this optimization process for rendering optical flow?

Third, I always encounter a  CUDA error after even 1 million frames or more, while this error never occurred when I only use RGB and depth as inputs. 
`RuntimeError: CUDA error: device-side assert triggered`
I guess that it may be caused by the optimization process in the second point (I'm trying to verify it) and may be caused by some NaN values. Maybe related to this problem, I wonder what's the difference between RGB and RGB-filled, and when I should use RGB-filled instead of RGB?

I'm looking forward to your reply and thanks so much!

UltronAI, 2021-09-18:
As an alternative method of calculating optical flow, the intrinsic matrix of the realsense camera in the simulator is helpful. If the above issues are difficult to fix, can you provide this info for a quick solution? 

Thanks!

roberto-martinmartin, 2022-05-12:
@fxia22 , has this been addressed in the newest versions? Should we close this pending issue?

fxia22, 2022-05-12:
This issue has been fixed.

closed 2022-05-12 by fxia22