StanfordVL/iGibson #128: Matterport Dataset floors.txt missing

elhamAm, 2021-09-16:
The original GibsonEnv codebase worked with Matterport but this newer version of the simulator requires a floors.txt for the environments. Is there a way to generate it or does your team have a copy of the required files? I know this issue has been resolved for the 3d2d dataset but it hasn't been resolved for the Matterport dataset

fxia22, 2021-09-17:
@elhamAm thanks for the question, can you follow this guide to generate `floors.txt` for matterport scans? https://github.com/StanfordVL/iGibson/tree/master/igibson/examples/gen_data

we will try to work with the matterport3d team to update a pre-processed version for iGibson.

fxia22, 2021-09-17:
@elhamAm we worked with matterport3d team and provided a pre-processed version. You can request `download_mp.py` again and download with `python2 download_mp.py --task_data igibson -o .`

ChengshuLi, 2021-09-17:
@elhamAm Could you please follow the new documentation we added here? 
http://svl.stanford.edu/igibson/docs/dataset.html#download-matterport3d-scenes
Thanks!

ChengshuLi, 2021-09-22:
Please feel free to re-open this issue if the problem persists.

closed 2021-09-22 by ChengshuLi