StanfordVL/iGibson #130: Calling Simulator fails with a keyError when ig_dataset folder is present

siodense, 2021-09-27:
Hi, So I installed iGibson via pip and am able to run demo_static just fine, but when I install the ig_dataset folder I get a keyError for 'enable_ag' when Simulator is called  

Here is the output from s = Simulator(mode="gui", image_width=700, image_height=700) from demo_static

INFO:root:Device 0 is available for rendering
INFO:root:Command '['/home/simono/anaconda3/envs/igibson/lib/python3.8/site-packages/igibson/render/mesh_renderer/build/test_device', '1']' returned non-zero exit status 1.
INFO:root:Device 1 is not available for rendering
INFO:root:Using device 0 for rendering
Compiling GLSL shader: /home/simono/anaconda3/envs/igibson/lib/python3.8/site-packages/igibson/render/mesh_renderer/shaders/450/equirect2cube_cs.glsl
Compiling GLSL shader: /home/simono/anaconda3/envs/igibson/lib/python3.8/site-packages/igibson/render/mesh_renderer/shaders/450/spmap_cs.glsl
Compiling GLSL shader: /home/simono/anaconda3/envs/igibson/lib/python3.8/site-packages/igibson/render/mesh_renderer/shaders/450/irmap_cs.glsl
Compiling GLSL shader: /home/simono/anaconda3/envs/igibson/lib/python3.8/site-packages/igibson/render/mesh_renderer/shaders/450/spbrdf_cs.glsl
Loading image: /home/simono/anaconda3/envs/igibson/lib/python3.8/site-packages/igibson/data/ig_dataset/scenes/background/photo_studio_01_2k.hdr
INFO: compiled pbr shaders
INFO:root:Loading /home/simono/anaconda3/envs/igibson/lib/python3.8/site-packages/igibson/data/assets/models/mjcf_primitives/cube.obj
INFO:root:Loading /home/simono/anaconda3/envs/igibson/lib/python3.8/site-packages/igibson/data/assets/models/mjcf_primitives/sphere8.obj
Traceback (most recent call last):
  File "/home/simono/anaconda3/envs/igibson/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/home/simono/anaconda3/envs/igibson/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/simono/anaconda3/envs/igibson/lib/python3.8/site-packages/igibson/examples/demo/motion_planning_example.py", line 41, in <module>
    run_example(args)
  File "/home/simono/anaconda3/envs/igibson/lib/python3.8/site-packages/igibson/examples/demo/motion_planning_example.py", line 12, in run_example
    nav_env = iGibsonEnv(
  File "/home/simono/anaconda3/envs/igibson/lib/python3.8/site-packages/igibson/envs/igibson_env.py", line 53, in __init__
    super(iGibsonEnv, self).__init__(
  File "/home/simono/anaconda3/envs/igibson/lib/python3.8/site-packages/igibson/envs/env_base.py", line 67, in __init__
    self.simulator = Simulator(
  File "/home/simono/anaconda3/envs/igibson/lib/python3.8/site-packages/igibson/simulator.py", line 169, in __init__
    self.gen_assisted_grasping_categories()
  File "/home/simono/anaconda3/envs/igibson/lib/python3.8/site-packages/igibson/simulator.py", line 1451, in gen_assisted_grasping_categories
    if v["enable_ag"]:
KeyError: 'enable_ag'

It seems to happen regardless of whether I install the iGibson2 scenes or iGibson1 scenes. Did I miss something in the installation process?

fxia22, 2021-09-27:
It looks like you have downloaded iGibson 1.0 dataset but used iGibson 2.0 code. Can you try to download the iGibson 2.0 dataset? 

you can remove the iGibson 1.0 dataset to avoid confusion. 

The instruction to download iGibson 2.0 dataset is here: http://svl.stanford.edu/igibson/docs/dataset.html#download-igibson-2-0-scenes-and-the-behavior-dataset-of-objects

siodense, 2021-09-28:
That was it. I had downloaded both and merged the folders absent mindedly so the folder which I thought contained the iGibson 2 data really contained iGibson 1. Thanks for the help!

closed 2021-09-28 by siodense