StanfordVL/iGibson #135: Can't Find Robot Action Space that Matches the iGibson Paper

mihdalal, 2021-11-11:
In the iGibson 1.0 paper, the following is described to be action space of the mobile manipulator:
The action space consists of four parts: a 1-
dimensional discrete value indicating whether to extend the
arm, 2-dimensional continuous values representing the robot
base’s linear and angular velocities, 6-dimensional continuous values representing the desired pose change of the endeffector, and 1-dimensional discrete value indicating whether
to open the gripper.
Where exactly is this implemented in the repository? 
The only thing I could find is FetchGripper which seems to have joint velocity control and continuous gripper control actions.

roberto-martinmartin, 2021-11-15:
Hi!
We are finishing a cleaner controller interface for the robots in iGibson that will be out in the next few days. That will have the right interface to choose between discrete open/close or continuous control of the gripper and also a more straightforward way to select between joint and Cartesian space. I can let you know when that is out.

Until then, you can take a look and/or use the IK controller we provide to control the robot in Cartesian space. There is an example to use it in examples/demo/ik_control_example.py

Let me know if that helps!

mihdalal, 2021-11-29:
Hi @roberto-martinmartin I just wanted to follow up on this - will the cleaner controller interface be out soon? It was been two weeks since your post so I wanted to check in. 
Thanks!

cgokmen, 2021-11-29:
We will cut a release this Friday. Thanks!

cgokmen, 2021-12-04:
Hi Murtaza, we are unfortunately running a little more behind schedule, but are close to releasing this functionality. Our current expectation is that the release will happen at some point during next week. We'll keep you posted.

roberto-martinmartin, 2022-01-07:
Hi @mihdalal ,
really sorry for the long delay. Winterbreak hit and delayed our (already delayed) plans. But everything should be available already! The action space has been cleaned up and boosted. You can select controllers for different components of the robots independently, depending on the robot's capabilities: base motion, end-effector motion, active vision (camera motion), and/or gripper control. Once the controllers have been specified, the action space is set. This covers the action space of our paper, but other useful additional variants, giving full control to the researchers.

We have a nice example but it is in our development code. Our releasing process to master is strict to avoid introducing bugs, but I have pushed it into a temporary branch for you to play and learn how to use. Check examples/robots/robot_control_example.py in this branch https://github.com/StanfordVL/iGibson/tree/cleanup-robot-demos. The functionalities are all in master branch, just the demo is not, but I thought it will help you getting started.

I hope this helps and I apologize again for the delay! Contact us again if you have other questions.

closed 2022-01-10 by roberto-martinmartin