StanfordVL/iGibson #136: How to download the Assets?

jshved, 2021-11-12:
I downloaded iGibson using the 3rd method from the source. So I continued to the Downloading Assets section with the first command:

`python -m igibson.utils.assets_utils --download_assets`

and it outputs that: /tmp/assets_igibson.tar.gz: No such file or directory

[image]

fxia22, 2021-11-12:
Oh right, in windows there is no `/tmp` folder. You can manually download from https://storage.googleapis.com/gibson_scenes/assets_igibson.tar.gz and decompress it to `igibson/data/assets`

jshved, 2021-11-12:
where then does the 
`python -m igibson.utils.assets_utils --download_demo_data`
folder suppose to go?
And how do I compile all of it on Windows?

mjlbach, 2021-11-12:
In the next release of iGibson, I've patched the downloaders to work on windows. You can read the source to see exactly what it is doing.

python -m igibson.utils.assets_utils --download_demo_data is just a helper that does:

```python
import igibson

tmp_file = os.path.join(tempfile.gettempdir(), "Rs.tar.gz")

if not os.path.exists(igibson.g_dataset_path):
    os.makedirs(igibson.g_dataset_path)

if not os.path.exists(os.path.join(igibson.g_dataset_path, "Rs")):
    os.system(
            "wget -c --retry-connrefused --tries=5 --timeout=5  "
            "https://storage.googleapis.com/gibson_scenes/Rs.tar.gz -O {}".format(tmp_file)
        )
    os.system("tar -zxf {} --directory {}".format(tmp_file, igibson.g_dataset_path))
```

You can try running the above file. As long as you have wget and tar on your path it will work.

closed 2021-11-12 by mjlbach