StanfordVL/iGibson #137: Numpy version doesn't work on M1 chip Mac

jshved, 2021-11-12:
Tried running 
` pip3 install numpy==1.21.0rc1 igibson`

Keep getting this error, are there any known solution? 
`clang: error: the clang compiler does not support 'faltivec', please use -maltivec and include altivec.h explicitly`

In numpy version 1.21 it fixes this error for the M1 chip, but when I use the fixed version, the configuration won't accept it.
 
`ERROR: Command errored out with exit status 1:
   command: /opt/homebrew/opt/python@3.9/bin/python3.9 /private/var/folders/wd/g2dnpjx90bj82wd8w3nks6vw0000gn/T/pip-standalone-pip-__wrj96f/__env_pip__.zip/pip install --ignore-installed --no-user --prefix /private/var/folders/wd/g2dnpjx90bj82wd8w3nks6vw0000gn/T/pip-build-env-typkzl4m/normal --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- 'numpy==1.14.5; python_version == "3.7"' 'numpy==1.19.3; python_version >= "3.9"' 'Cython>=0.29; python_version < "3.8"' 'numpy==1.17.5; python_version == "3.8"' 'Cython>=0.29.14; python_version == "3.8"' 'Cython>=0.29.15; python_version >= "3.9"' pkgconfig
       cwd: None
  Complete output (912 lines):
  Ignoring numpy: markers 'python_version == "3.7"' don't match your environment
  Ignoring Cython: markers 'python_version < "3.8"' don't match your environment
  Ignoring numpy: markers 'python_version == "3.8"' don't match your environment
  Ignoring Cython: markers 'python_version == "3.8"' don't match your environment`

ChengshuLi, 2021-11-13:
Could you try an older version of python (say py36?) and an older version
of numpy (say 1.1x?)

jshved, 2021-11-13:
I tried it still didn't work.

closed 2021-11-13 by jshved

EvaSDK, 2022-02-04:
Could you share if you managed to find a workaround for this ? I am trying to build wheel packages for our internal index so coworkers do not have to setup everything by themselves.

mjlbach, 2022-02-04:
```bash
# If you don't have conda-forge enabled
conda config --add channels conda-forge
conda config --set channel_priority strict 

# Make a new iGibson environment
conda create -n igibson python=3.8 pip
conda activate igibson

# Install iGibson
git clone git@github.com:stanfordvl/iGibson.git
cd iGibson
pip install -e .

# Run iGibson (you'll need to adjust these paths)
export GIBSON_ASSETS_PATH=$HOME/Documents/ig_data/assets                                                                                                           
export IGIBSON_DATASET_PATH=$HOME/Documents/ig_data/ig_dataset 
python igibson/examples/renderer/mesh_renderer_pbr_example.py  
```

The numpy version installed by the pip solver (1.22.2) works fine on my m1 mac.

Here's the exact requirements.txt

```
aenum==3.1.8
appnope==0.1.2
asttokens==2.0.5
attrs==21.4.0
backcall==0.2.0
bddl==0.0.5
black==22.1.0
click==8.0.3
cloudpickle==2.0.0
cycler==0.11.0
decorator==5.1.1
executing==0.8.2
fonttools==4.29.1
freetype-py==2.2.0
future==0.18.2
gitdb==4.0.9
GitPython==3.1.26
GPUtil==1.4.0
gym==0.21.0
h5py==3.6.0
importlib-resources==5.4.0
iniconfig==1.1.1
ipython==8.0.1
ipython-genutils==0.2.0
jedi==0.18.1
jsonschema==4.4.0
jupyter-core==4.9.1
jupytext==1.13.6
kiwisolver==1.3.2
markdown-it-py==1.1.0
matplotlib==3.5.1
matplotlib-inline==0.1.3
mdit-py-plugins==0.3.0
mypy-extensions==0.4.3
nbformat==5.1.3
networkx==2.6.3
numpy==1.22.2
opencv-python==4.5.5.62
packaging==21.3
pandas==1.4.0
parso==0.8.3
pathspec==0.9.0
pexpect==4.8.0
pickleshare==0.7.5
Pillow==9.0.1
platformdirs==2.4.1
pluggy==1.0.0
progressbar==2.5
prompt-toolkit==3.0.26
ptyprocess==0.7.0
pure-eval==0.2.2
py==1.11.0
py360convert==0.1.0
pybullet-svl==3.1.6.1
Pygments==2.11.2
pyparsing==3.0.7
pyrsistent==0.18.1
pytest==7.0.0
python-dateutil==2.8.2
pytz==2021.3
PyYAML==6.0
scipy==1.7.3
six==1.16.0
smmap==5.0.0
stack-data==0.1.4
toml==0.10.2
tomli==2.0.0
tqdm==4.62.3
traitlets==5.1.1
transforms3d==0.3.1
trimesh==3.9.43
typing_extensions==4.0.1
urllib3==1.26.8
wcwidth==0.2.5
zipp==3.7.0
```