StanfordVL/iGibson #139: Quickstart Example Launches then Crashes

jshved, 2021-11-16:
Hey I am a windows user. I used the compiling from source method and when I went to run this command:
`python -m igibson.examples.demo.env_example`

I got this error right after the simulation launched then crashed. I don't quite understand it because I downloaded the assets and demo data that were recommended.
`BDDL could not be imported - object taxonomy / abilities will be unavailable.
torch is not available, falling back to rendering to memory(instead of tensor)
INFO: Succesfully initialized GLFW context and window!
WARNING:root:Environment texture not available, cannot use PBR.
starting thread 0
started testThreads thread 0 with threadHandle 0000000000002C14
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
Version = 4.6.14830 Compatibility Profile Context 21.10.20 27.20.21020.3
Vendor = ATI Technologies Inc.
Renderer = AMD Radeon(TM) Graphics
b3Printf: Selected demo: Physics Server
starting thread 0
started MotionThreads thread 0 with threadHandle 0000000000003CD4
MotionThreadFunc thread started
INFO:root:Loading /data/assets\models/mjcf_primitives/cube.obj
ERROR:root:Warning:
ERROR:root:Error: Cannot open file [/data/assets\models/mjcf_primitives/cube.obj]

ERROR:root:Failed to load: /data/assets\models/mjcf_primitives/cube.obj
numActiveThreads = 0
stopping threads
Thread with taskId 0 with handle 0000000000003CD4 exiting
Thread TERMINATED
finished
numActiveThreads = 0
btShutDownExampleBrowser stopping threads
Thread with taskId 0 with handle 0000000000002C14 exiting
Thread TERMINATED`

mjlbach, 2021-11-16:
Seems like you are missing our assets bundle. How did you download the assets?

jshved, 2021-11-16:
I downloaded the zip and unzipped it into assets/ folder. I am not missing the assets, it just can't open them.

roberto-martinmartin, 2021-11-16:
Could you confirm the file exists in the printed path: `your_ig_installation_folder/igibson/data/assets/models/mjcf_primitives/cube.obj`?

I wonder about the "\" in the path, but I do not program on Windows, I'm not sure if that is expected

also, there are some variables for the datasets, could you tell us the content of your `igibson/global_config.yaml`
Thanks!

jshved, 2021-11-16:
Yep, it exists. I tried changing the "\" to "/"  and back and forth, but it made no difference. I also tried modifying the path with no success.

the contents of my global_config.yaml is:

assets_path: /data/assets # put either absolute path or relative to current dirrectory
g_dataset_path: /data/g_dataset
ig_dataset_path: /data/ig_dataset
threedfront_dataset_path: /data/threedfront_dataset
cubicasa_dataset_path: /data/cubicasa_dataset
key_path: /data/igibson.key

mjlbach, 2021-11-16:
I was not able to reproduce, here are the exact steps I did on a clean Windows 10 system with nvidia drivers, conda, + steamvr (not necessary but those were the two things I installed):

1. `git clone https://github.com/stanfordvl/igibson.git --recursive`
2. `conda create -n igibson python=3.8`
3. `conda activate igibson`
4. `cd igibson`
5. `pip install -e .`
6. `curl -fLO https://storage.googleapis.com/gibson_scenes/assets_igibson.tar.gz -O assets_igibson.tar.gz`
7. `tar xzvf ./assets_igibson.tar.gz`
8. `mkdir igibson/data`
9. `mv assets igibson/data`
10. `curl -fLO https://storage.googleapis.com/gibson_scenes/Rs.tar.g`
11. `tar xzvf ./Rs.tar.gz`
12. `mkdir -p .\igibson\data\g_dataset`
13. `mv Rs igibson/data/g_dataset`

roberto-martinmartin, 2021-11-16:
why did you add the "//" at the beginning of the paths in your `global_config.yaml`? At least in Linux that would mean the folder is /data in the root directory, not relative, right? what happens if you remove it? or you change it to your full path: /full/path/in/your/directory/data/assets?

jshved, 2021-11-16:
thank you, you're a genius.

closed 2021-11-16 by jshved