StanfordVL/iGibson #140: Loading an external 3D model to iGibson

asfandasfo, 2021-11-18:
Hi, I was trying to load my own 3D model. The original model is in .blender format. I exported it to .obj format and followed the instructions [here](https://github.com/StanfordVL/iGibson/tree/master/igibson/examples/gen_data) but couldn't generate the floor.txt.

roberto-martinmartin, 2021-11-18:
what is the error?

asfandasfo, 2021-11-19:
> what is the error?

@roberto-martinmartin 
[image]

roberto-martinmartin, 2021-11-20:
Looking at the code of the assert, it looks like your mesh has more than 2 faces (triangles) connected to the same edge. Sounds like a malformed mesh. The library we use to process the meshes (meshcut) requires edges to connect max 2 triangles. Maybe you can clean the input mesh before processing it with our script to remove those cases.

this is the code from meshcut that triggers the asset:
```
# Sanity check : max 2 faces per edge
for e, tris in self.edges_to_tris.items():
    assert len(tris) <= 2
```

asfandasfo, 2021-12-03:
@roberto-martinmartin I tried this with matterport scene(17DRP5sb8fy) as well and got the same assertion error. After cleaning the input, I'm getting this.
[image]

fxia22, 2021-12-03:
You can try commenting out the assertion in meshcut code.

asfandasfo, 2021-12-07:
> You can try commenting out the assertion in meshcut code.
> […](#)

Still same error...

https://user-images.githubusercontent.com/11736209/144590660-ec14dbc2-fa03-4db3-a074-a22539f2ff69.png

asfandasfo, 2021-12-07:
@fxia22 @roberto-martinmartin 

Thanks, I am able to solve this issue with matterport scene but still struggling loading my own model.

roberto-martinmartin, 2022-01-07:
let us know if you have further questions but it looks like a mesh formatting issue. I'll close the issue, feel free to reopen with more details

closed 2022-01-07 by roberto-martinmartin

asfandasfo, 2022-01-10:
> let us know if you have further questions but it looks like a mesh formatting issue. I'll close the issue, feel free to reopen with more details

Hi, I was managed to load my own model without errors but I don't see anything in pybullet. When I checked the external view I see my model loaded inside a different Gibson scene with weird texture. Any suggestions on what could be the problem?
[image]