StanfordVL/iGibson #141: Installation Issue on Windows10

Sophycountry, 2021-12-01:
I have tried to install iGibson 2.0 under Windows10 environment using 3rd method, but it will always fail at "cmake -- version".
Below is the error message and I have already installed cmake==3.22.0 with pip/conda(both tried) before in my conda virtual env.

ERROR: Command errored out with exit status 1:
     command: 'C:\Users\songf\anaconda3\envs\igibson\python.exe' -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\songf\\Documents\\program\\iGibson\\setup.py'"'"'; __file__='"'"'C:\\Users\\songf\\Documents\\program\\iGibson\\setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' develop --no-deps
         cwd: C:\Users\songf\Documents\program\iGibson\
    Complete output (49 lines):
    running develop
    running egg_info
    writing igibson.egg-info\PKG-INFO
    writing dependency_links to igibson.egg-info\dependency_links.txt
    writing requirements to igibson.egg-info\requires.txt
    writing top-level names to igibson.egg-info\top_level.txt
    reading manifest file 'igibson.egg-info\SOURCES.txt'
    reading manifest template 'MANIFEST.in'
    adding license file 'LICENSE'
    writing manifest file 'igibson.egg-info\SOURCES.txt'
    running build_ext
    Traceback (most recent call last):
      File "C:\Users\songf\anaconda3\envs\igibson\lib\runpy.py", line 194, in _run_module_as_main
        return _run_code(code, main_globals, None,
      File "C:\Users\songf\anaconda3\envs\igibson\lib\runpy.py", line 87, in _run_code
        exec(code, run_globals)
      File "C:\Users\songf\anaconda3\envs\igibson\Scripts\cMake.exe\__main__.py", line 4, in <module>
    ModuleNotFoundError: No module named 'cmake'
    C:\Users\songf\AppData\Local\Temp\pip-build-env-mzyh__cf\overlay\Lib\site-packages\setuptools\command\easy_install.py:156: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
      warnings.warn(
    C:\Users\songf\AppData\Local\Temp\pip-build-env-mzyh__cf\overlay\Lib\site-packages\setuptools\command\install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
      warnings.warn(
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\songf\Documents\program\iGibson\setup.py", line 131, in <module>
        setup(
      File "C:\Users\songf\AppData\Local\Temp\pip-build-env-mzyh__cf\overlay\Lib\site-packages\setuptools\__init__.py", line 153, in setup
        return distutils.core.setup(**attrs)
      File "C:\Users\songf\anaconda3\envs\igibson\lib\distutils\core.py", line 148, in setup
        dist.run_commands()
      File "C:\Users\songf\anaconda3\envs\igibson\lib\distutils\dist.py", line 966, in run_commands
        self.run_command(cmd)
      File "C:\Users\songf\anaconda3\envs\igibson\lib\distutils\dist.py", line 985, in run_command
        cmd_obj.run()
      File "C:\Users\songf\AppData\Local\Temp\pip-build-env-mzyh__cf\overlay\Lib\site-packages\setuptools\command\develop.py", line 34, in run
        self.install_for_development()
      File "C:\Users\songf\AppData\Local\Temp\pip-build-env-mzyh__cf\overlay\Lib\site-packages\setuptools\command\develop.py", line 114, in install_for_development
        self.run_command('build_ext')
      File "C:\Users\songf\anaconda3\envs\igibson\lib\distutils\cmd.py", line 313, in run_command
        self.distribution.run_command(command)
      File "C:\Users\songf\anaconda3\envs\igibson\lib\distutils\dist.py", line 985, in run_command
        cmd_obj.run()
      File "C:\Users\songf\Documents\program\iGibson\setup.py", line 45, in run
        out = subprocess.check_output(["cMake", "--version"])
      File "C:\Users\songf\anaconda3\envs\igibson\lib\subprocess.py", line 415, in check_output
        return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
      File "C:\Users\songf\anaconda3\envs\igibson\lib\subprocess.py", line 516, in run
        raise CalledProcessError(retcode, process.args,
    subprocess.CalledProcessError: Command '['cMake', '--version']' returned non-zero exit status 1.
    ----------------------------------------
ERROR: Command errored out with exit status 1: 'C:\Users\songf\anaconda3\envs\igibson\python.exe' -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\songf\\Documents\\program\\iGibson\\setup.py'"'"'; __file__='"'"'C:\\Users\\songf\\Documents\\program\\iGibson\\setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' develop --no-deps Check the logs for full command output.

Sophycountry, 2021-12-01:
I have tried the following procedure:
1. pip install cmake
2. conda install cmake
3. install cmake on my computer from the cmake webset
"cmake --version" can work on cmd, and "import cmake" can success on python.
Thank you for answer!

Sophycountry, 2021-12-02:
I have solved the problem.... it's an error about "git clone", that the submodles not git successfully such as cryptopp and so on.

closed 2021-12-02 by Sophycountry

ToddBear, 2022-01-24:
Hi Sophycountry, i met the same problem when installed it in windows. I wonder how you solved it ?

1415723621, 2022-03-20:
@ToddBear Hello,I met the same problem too and i solved it by twe step.First,i found git ----recursive failed to install subgit folder,so i manually git the subfolder in /render/(crytopp glfw glm and pybind11 ).Then it can`t find the cmake reboot cause there is mutil version cmake lib(sudo apt get install , pip install cmake,conda install and so so),so i pip uninstall and conda uninstall surplus cmake version and just leave the right one, finally it works.