StanfordVL/iGibson #142: File missng

meier-johannes94, 2021-12-01:
I tried to run BehaviorChallenge with discrete actions, so I made some changes and now I am receiving the following error message: 
```
ERROR:root:Failed to load: /is/sg2/jmeier/behaviorchallenge/iGibson/igibson/data/assets/models/mjcf_primitives/sphere8.obj
```
I do not even have a folder `assets` in `data`. I only have the key and `ig_dataset` there.

roberto-martinmartin, 2021-12-01:
Hi Johannes,

you need to download the assets. This includes the basic models for simulation (robots, shape primitives...). The ig_dataset includes object and scene models and is necessary as well for BEHAVIOR.

See the instructions for the assets here (https://stanfordvl.github.io/behavior/installation.html), Option 1, step 6. In case this fails for you (we had some issues with Windows users in the past), you can download and decompress the assets manually. Follow the instructions here (http://svl.stanford.edu/igibson/docs/installation.html#downloading-the-assets).

Let me know if this fixes your issues.

roberto-martinmartin, 2021-12-01:
Additional information about the different datasets available for BEHAVIOR in iGibson can be found here: http://svl.stanford.edu/igibson/docs/dataset.html

closed 2021-12-01 by meier-johannes94