StanfordVL/iGibson #143: Two questions: Discrete action space + Global map creation

meier-johannes94, 2021-12-01:
Thanks for your recent help. This was very helpful and very quick! 

I have a few more questions: 
- Do you have an example or minimal working example for the discrete action space case? I tried it myself, but currently I am getting the following error. 
```No inertial data for link, using mass=1, localinertiadiagonal = 1,1,1, identity local inertial frameb3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:
eyeprocessing eye
INFO:root:Loading /is/sg2/jmeier/behaviorchallenge/iGibson/igibson/data/assets/models/vr_agent/vr_eye/normal_color/vr_head.obj
Traceback (most recent call last):
  File "stable_baselines3_behavior_example.py", line 175, in <module>
    main()
  File "stable_baselines3_behavior_example.py", line 126, in main
    env = make_env(0)()
  File "stable_baselines3_behavior_example.py", line 118, in _init
    physics_timestep=1 / 300.0,
  File "/is/sg2/jmeier/behaviorchallenge/iGibson/igibson/envs/behavior_mp_env.py", line 103, in __init__
    automatic_reset=automatic_reset,
  File "/is/sg2/jmeier/behaviorchallenge/iGibson/igibson/envs/behavior_env.py", line 64, in __init__
    render_to_tensor=render_to_tensor,
  File "/is/sg2/jmeier/behaviorchallenge/iGibson/igibson/envs/igibson_env.py", line 60, in __init__
    render_to_tensor=render_to_tensor,
  File "/is/sg2/jmeier/behaviorchallenge/iGibson/igibson/envs/env_base.py", line 78, in __init__
    self.load()
  File "/is/sg2/jmeier/behaviorchallenge/iGibson/igibson/envs/behavior_env.py", line 177, in load
    self.load_action_space()
  File "/is/sg2/jmeier/behaviorchallenge/iGibson/igibson/envs/behavior_mp_env.py", line 111, in load_action_space
    if self.activity_relevant_objects_only:
AttributeError: 'BehaviorMPEnv' object has no attribute 'activity_relevant_objects_only'
```
- I would like to construct a discrete global map of the living room. Like if you take a birds eye view I would like to know where is a wall, where is a "fixed" object, where are the tiny objects, etc. Can you give me some hints, where in the code to look for this? I assume this is not a default functionality yet. 

I appreciate your help!

cgokmen, 2021-12-03:
Hi Johannes,

Thanks for your interest. A minimal working example of the discrete action space will be available in the upcoming few weeks (we're in the process of cleaning up these interfaces so there's a chance some of the examples might break like mentioned above). We're trying to get these out as soon as possible.

As for the discrete map, there is working code for a **room_based** a segmentation in our scene implementation: https://github.com/StanfordVL/iGibson/blob/c6854f11eec5d935fa3ef3d6d4852c6571beab4b/igibson/scenes/igibson_indoor_scene.py#L991-L1007 The `world_to_seg_map` utility function provides you with a ready-to-use mechanism for converting world-to-image coordinates. If you want to generate a w by h image, you can iterate through all w*h pixels, find corresponding world coordinates, use pybullet's rayTest function (documentation [here](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#)) to cast top-down rays to get the ID of the object at that spot, and finally get the object's category by looking it up in the scene: `scene.objects_by_id[ray_cast_hit_body_id].category`.

Note that our renderer already has a semantic segmentation mode so you could technically use it to get such an image too, you would just need to correctly set its P and V matrices to correctly position the camera and set its perspective. I'm less familiar with how you can do this and get an image without perspective warps but perhaps @fxia22 could help.

Let us know if we can help any further!
Cem

roberto-martinmartin, 2022-01-07:
there is an example on how to generate a top down view (examples/observations/generate_topdown_semgeg.py in the new release). That would be a good starting point. A problem is that our rendering uses a pinhole camera model. If you want an orthographic projection, you probably would need to write your own renderer/shader. A place to look would be our code for the shadow generation that uses that type of projection

closed 2022-01-07 by roberto-martinmartin