StanfordVL/iGibson #144: 2 Questions for iGibson 2.0 / Bheavior Challenge

meier-johannes94, 2021-12-07:
Thanks, your recent help was great! I am amazed by your support, thank you! 

Here a few more points: 

- I tried to use a different activity. Therefore I changed `behavior_onboard_sensing.yaml` by setting `task: boxing_books_up_for_storage`, but then I got an error message that `...fixed_furniture` file can't be found. So I activated `online_sampling`. in the yaml-file. Does this randomize which objects are loaded and where they are placed? 

But then I got: 
```
Traceback (most recent call last):
  File "stable_baselines3_behavior_example.py", line 202, in <module>
    main()
  File "stable_baselines3_behavior_example.py", line 137, in main
    env = make_env(0)()
  File "stable_baselines3_behavior_example.py", line 129, in _init
    physics_timestep=1 / 300.0,
  File "/is/sg2/jmeier/behaviorchallenge/iGibson/igibson/envs/behavior_mp_env.py", line 108, in __init__
    automatic_reset=automatic_reset,
  File "/is/sg2/jmeier/behaviorchallenge/iGibson/igibson/envs/behavior_env.py", line 64, in __init__
    render_to_tensor=render_to_tensor,
  File "/is/sg2/jmeier/behaviorchallenge/iGibson/igibson/envs/igibson_env.py", line 60, in __init__
    render_to_tensor=render_to_tensor,
  File "/is/sg2/jmeier/behaviorchallenge/iGibson/igibson/envs/env_base.py", line 78, in __init__
    self.load()
  File "/is/sg2/jmeier/behaviorchallenge/iGibson/igibson/envs/behavior_env.py", line 175, in load
    self.load_task_setup()
  File "/is/sg2/jmeier/behaviorchallenge/iGibson/igibson/envs/behavior_env.py", line 164, in load_task_setup
    self.load_behavior_task_setup()
  File "/is/sg2/jmeier/behaviorchallenge/iGibson/igibson/envs/behavior_env.py", line 132, in load_behavior_task_setup
    online_sampling=online_sampling,
  File "/is/sg2/jmeier/behaviorchallenge/iGibson/igibson/activity/activity_base.py", line 92, in initialize_simulator
    self.initial_state = self.save_scene()
  File "/is/sg2/jmeier/behaviorchallenge/iGibson/igibson/activity/activity_base.py", line 98, in save_scene
    self.state_history[snapshot_id] = save_internal_states(self.simulator)
  File "/is/sg2/jmeier/behaviorchallenge/iGibson/igibson/utils/checkpoint_utils.py", line 38, in save_internal_states
    for name, obj in simulator.scene.objects_by_name.items():
AttributeError: 'NoneType' object has no attribute 'objects_by_name'
```
- I would like to use the editor of behavior challenge to create a custom activity, but it seems not accessible. Can you already say, when we can use it again? https://behavior-annotations.herokuapp.com/. If the support for BehaviorChallenge is as quick as yours, I also don't mind to post it there ;-)

sanjanasrivastava, 2021-12-07:
Hi! Regarding creating a custom activity - the link you tried to access is deprecated. You can use this link: https://behavior.stanford.edu/activity-annotation. It's a quick instruction sheet. 

At the bottom, you'll see "Click here to access the activity annotator", which will take you to an accessible version of the interface!

meier-johannes94, 2021-12-07:
@sanjanasrivastava 
I found the link in the main menu here: http://svl.stanford.edu/behavior/index.html
Is there a more elegant way to integrate manually created activities than just copying it into `activity_definitions`? I just wonder, because bddl seems to be a separate package.

sanjanasrivastava, 2021-12-07:
@meier-johannes94 are you asking about using the custom activity yourself, or about integrating it with the public version of `bddl`? If the former, then yes you just copy it in the format you see - we don't have another way at the moment to generate a file from the website. Specifically put it at `activity_definitions/<your_activity_name>/problem<your_unique_activity_instance_number>.bddl`. 

If you are asking about adding your activity to the public version of `bddl`, then you will have to 1) verify that your activity can be sampled in an iGibson scene 2) make a pull request to the main `bddl` repo.

meier-johannes94, 2021-12-07:
@sanjanasrivastava 
If I just use private for now, that is okay :-) Thanks! 
And do I have to look for anything else besides putting in the order? Because when I switched the task in the `yaml` config-file I got above's error message.

cgokmen, 2021-12-07:
Hi Johannes,

* Re: the link, you've actually found an outdated website (http://svl.stanford.edu/behavior/index.html) that we haven't redirected to the new one yet. You can check out our new website at https://behavior.stanford.edu, and we just updated the URL you linked above to redirect to the new one. Thanks for pointing this out.
* Re: the custom activity definition's location, you're right that iGibson currently does not provide a nice entrypoint for using your custom activity definitions. You can either get BDDL from the StanfordVL/BDDL repo and `pip install -e .` to install it (remove your existing BDDL first by doing pip uninstall) and add your definition in the activity definitions directory, or you can change code within iGibson to make use of [this feature](https://github.com/StanfordVL/bddl/blob/8646ff5678e3956411579ddfc9ea25df29b9ed36/bddl/parsing.py#L151-L157) in BDDL that lets you enter your own task definition. I have added it to our roadmap to make it possible to use custom activity definitions from the iGibson config yaml files.
* Re: your original question, online_sampling does indeed enable randomization and it's a slow process. @ChengshuLi can comment on the particular bug you saw when you enabled that, but I think you _don't_ want to enable that in the first place. The reason why you're seeing the error with the fixed_furniture file when you don't enable online sampling is because we have cached scenes for only a subset of all possible task-scene combinations. You can go inside the igibson/data/ig_dataset/scenes directory and search for filenames that contain the activity you want to find which scenes it's available in.
* Re: carrying with containers - it's definitely possible & encouraged! The currently available discrete action implementation does not support this however, although we are aiming to release an improved discrete action system that has support for this by the end of the year. You can still use it with continuous actions.
* It's not yet possible to do all 100 activities with the discrete actions as you mention: cleaning tasks are not possible and neither is toggling. The aforementioned discrete actions update will get us closer.

Let us know if you need anything else -
Cem

meier-johannes94, 2021-12-07:
@cgokmen 
Hi Cem, wow, your support is really great! Fast and high quality!

`Re: your original question, online_sampling does indeed enable randomization and it's a slow process. @ChengshuLi can comment on the particular bug you saw when you enabled that, but I think you don't want to enable that in the first place. The reason why you're seeing the error with the fixed_furniture file when you don't enable online sampling is because we have cached scenes for only a subset of all possible task-scene combinations. You can go inside the igibson/data/ig_dataset/scenes directory and search for filenames that contain the activity you want to find which scenes it's available in.`

OK, I found them. And I think I will have to create some custom scenes. In that case I will have to enable randomization, otherwise it doesn't work, right? 

What is actually the idea behind randomization? Is it like we need to find suitable positions for all the objects s.t. they can be reached? Does it mean that if randomization is disabled that the scene always looks 100% same? Does it mean that all the furniture, the robot's position and the object's position is always the same? If so, don't you worry that it would be better to have more diversity? E.g. for `boxing_books_up_for_storage` I found 30 `fixed_furniture`. If I say we make a train/test split of 75/25, then we have about 7(8) for testing and 23(22) for training. 

About what extent of performance drop are we talking about? And it is just for loading the scene, right? Once it is loaded, the performance is still the same? 

Is it possible to create them somehow myself?

meier-johannes94, 2021-12-08:
@cgokmen
One additional probably very simple question for you. What is the difference between:
- open & drop
- close & grasp
? 
And in what order would I apply the actions?

cgokmen, 2021-12-08:
Re: the scene stuff: the "sampling" process randomizes the scene by putting task-relevant objects at different (eligible) locations. The scene furniture itself doesn't change, but the small objects and robots will be placed differently, with different furniture objects etc. being picked for the different task conditions too. We don't enable randomization at runtime usually since it takes a bit to run, but instead we batch-generate randomized copies of the scene (where the objects and robots will be placed differently), and then load them later with randomization disabled. Additional diversity can come from using different scenes (which have different floorplans/furniture) and enabling texture randomization.

So anyway, you can sample once and cache the results, then load that like a regular pre-cached iGibson scene. @ChengshuLi can help you with how - we will release documentation soon but he can maybe give you some quick tips about the entrypoint.

Re: the discrete actions:
Open & close are used on objects with the `openable` ability annotation, like doors and drawers. They will cause the object to be opened/closed (which is defined w.r.t. the object's joints).

Grasp/release involves the agent grasping or releasing an object. So they are very different actions.

I should warn you however that the performance of the currently-available version of the primitives is rather poor, and they don't work through physics simulation but instead "magically" (e.g. things just get teleported). So as you test, if you notice that the performance is bad - might be worth waiting around a bit for the new version.

cgokmen, 2021-12-08:
And yes, regardless of randomization, the performance is the same after loading the scene. But the loading process can significantly longer.

mjlbach, 2021-12-08:
> And yes, regardless of randomization, the performance is the same after loading the scene. But the loading process can significantly longer.

If you do online sampling, we cannot merge fixed links since we need to query individual links during the sampling. This is a performance optimization that is required to accommodate the large number of scene objects, so there will be a runtime cost.

meier-johannes94, 2021-12-08:
@cgokmen
You said the new version with the speed-up might be released around end of year, right? 

`I should warn you however that the performance of the currently-available version of the primitives is rather poor, and they don't work through physics simulation but instead "magically" (e.g. things just get teleported). So as you test, if you notice that the performance is bad - might be worth waiting around a bit for the new version.`
Having a first look at the code, it seems as if it is still planning the route the robot would go if we had continuous action (at least for navigation). So would it theoretically be possible to modify the `navigate_to` process such that we let it still plan the full route but then only let it execute a fraction of the whole route? E.g. if we want to navigate to object A, subdivide the action that is usually taken in one timestep into 10 timesteps? I know this is not what your intention has been so far, but do you think this is possible with reasonable effort?

meier-johannes94, 2021-12-09:
> @cgokmen You said the new version with the speed-up might be released around end of year, right?
> 
> `I should warn you however that the performance of the currently-available version of the primitives is rather poor, and they don't work through physics simulation but instead "magically" (e.g. things just get teleported). So as you test, if you notice that the performance is bad - might be worth waiting around a bit for the new version.` Having a first look at the code, it seems as if it is still planning the route the robot would go if we had continuous action (at least for navigation). So would it theoretically be possible to modify the `navigate_to` process such that we let it still plan the full route but then only let it execute a fraction of the whole route? E.g. if we want to navigate to object A, subdivide the action that is usually taken in one timestep into 10 timesteps s.t. the agent would have to call `step(action)` 10 times before we arrive? I know this is not what your intention has been so far, but do you think this is possible with reasonable effort?

OK, I found the answer for this myself.

ChengshuLi, 2022-06-10:
Feel free to re-open the issue if you have further questions.

closed 2022-06-10 by ChengshuLi