StanfordVL/iGibson #147: change scene without changing the robot

DanielaPamplona, 2021-12-17:
Hi,
I was wondering if without using the environment class, I can run the simulator and, at some point, change the scene without changing the robot, I want to keep the same robot instance. 

I used the reload function of the simulator class, and I successfully import the new scene, but when I try to import the old robot it gives an assertion error.

The code (based on scene_interactive_object_rand_example.py):

##########################################

from igibson.robots.turtlebot_robot import Turtlebot
from igibson.simulator import Simulator
from igibson.scenes.igibson_indoor_scene import InteractiveIndoorScene
from igibson.objects.ycb_object import YCBObject
from igibson.utils.utils import parse_config
from igibson.render.mesh_renderer.mesh_renderer_settings import MeshRendererSettings
from igibson.render.profiler import Profiler
import igibson
import os

def main():
    config = parse_config(os.path.join(igibson.example_config_path, 'turtlebot_demo.yaml'))
    settings = MeshRendererSettings(enable_shadow=False, msaa=False)
    sim = Simulator(mode='headless', image_width=256,image_height=256, rendering_settings=settings)

    scene = InteractiveIndoorScene('Rs_int',
                              build_graph=True,
                              pybullet_load_texture=True)
    sim.import_ig_scene(scene)
    turtlebot = Turtlebot(config)
    sim.import_robot(turtlebot)

    for _ in range(10):
        obj = YCBObject('003_cracker_box')
        sim.import_object(obj)
        obj.set_position_orientation(np.random.uniform(
            low=0, high=2, size=3), [0, 0, 0, 1])

    print(sim.renderer.instances)

    for i in range(100):
        with Profiler('Simulator step'):
            turtlebot.apply_action([0.1, 0.1])
            sim.step()
            rgb = sim.renderer.render_robot_cameras(modes=('rgb'))
   
   
    print("changing scene")
    sim.reload()
    scene = InteractiveIndoorScene('Merom_0_int', build_graph=True,pybullet_load_texture=True)#Wainscott_0_int
    sim.import_ig_scene(scene)
    sim.import_robot(turtlebot) 

    for _ in range(10):
        obj = YCBObject('003_cracker_box')
        sim.import_object(obj)
        obj.set_position_orientation(np.random.uniform(
            low=0, high=2, size=3), [0, 0, 0, 1])

    print(sim.renderer.instances)

    for i in range(100):
        with Profiler('Simulator step'):
            turtlebot.apply_action([0.1, 0.1])
            sim.step()
            rgb = sim.renderer.render_robot_cameras(modes=('rgb'))
        pdb.set_trace() 
    sim.disconnect()

if __name__ == '__main__':
    main()

####################

And the error

####################

  File "simulator_interactive_various_scenes.py", line 65, in <module>
    main()
  File "simulator_interactive_various_scenes.py", line 45, in main
    sim.import_robot(turtlebot) 
  File "/home/louisette/.local/lib/python3.8/site-packages/igibson/simulator.py", line 267, in wrapped_load_func
    res = load_func(*args, **kwargs)
  File "/home/louisette/.local/lib/python3.8/site-packages/igibson/simulator.py", line 913, in import_robot
    state.initialize(self)
  File "", line 51, in initialize
    assert not self._initialized, "State is already initialized."
AssertionError: State is already initialized.

##########################

I understand that the object was already initialized and therefore it cannot be imported again. But I cannot see another way of doing this.

Any ideas?

cgokmen, 2021-12-17:
When you reload the sim, your original robot gets deleted along with everything else in the simulator. That means not only is the body not where you expect it to be anymore, but it's not even loaded into the physics engine.

Overall, we don't support using the same robot instance across different simulator loads.

But I wonder what you are trying to do this? Perhaps we can find a workaround.

DanielaPamplona, 2021-12-17:
Thanks for replying so fast!
Actually, I just want to change the scene where the robot is. 
Say something like there is a village of houses and the same robot goes from one house to the other after cleaning up.
My initial solution was to import the scenes sequentially in the same simulator load, but I did not managed to make it work.
Any suggestion?

cgokmen, 2021-12-17:
You can create another robot with the same config in your new scene. It shouldn't be too important that it's not the _same_ robot but another robot of the same type with the same configuration right? Or is there some sort of state that you want to retain across the scene change? Perhaps there's some way we can copy over that state.

DanielaPamplona, 2021-12-18:
I will try that, but now I will go on vacation. Please keep this issue open, I will come back with some feedback.
Tks!

DanielaPamplona, 2022-01-20:
Hi,
To create another robot works. Overall, this solution of reloading the simulator, loading a new scene and a new robot, takes a lot of time (and it is very verbosed!), but it works. If you have any other ideas, I am open to cleaner solutions. 
Anyway, thanks! 
You can close the issue.

closed 2022-06-10 by ChengshuLi