StanfordVL/iGibson #148: Unified pixel coordinate scheme

sycz00, 2021-12-22:
Dear iGibson Team,

First things first: thanks for maintaining the lib, its amazing !

My question:
I am using iGibson to train an agent exploring the environment given some top down map of the environment. The map is successively revealed during the episode by the agent itself based on depth and semantic segmentation mask. Now I want to increase the generalization of the method by training on different environments. 
I use some naive method for converting the world coordinates into pixel coordinates for the map as such : 
`x = (x_world+self.grid_offset[0])/self.grid_spacing[0]
 y = (y_world+self.grid_offset[1])/self.grid_spacing[1]
 return [np.round(x)+self.offset,np.round(y)+self.offset]`

Now the world coordinates are distinct from each other depending on the scene used. My question is if there exist any unified method such that I do not need to modify these spacing's for each scene. I am even required to resize the maps depending on the scene.

cgokmen, 2022-01-06:
This here is our only world-map coordinate conversion that iGibson includes:
https://github.com/StanfordVL/iGibson/blob/c0dba21952eb8a9323278be276ea9295425a1f7c/igibson/scenes/igibson_indoor_scene.py#L960-L977

Obviously this will have a different result based on the seg map's resolution. I'm not sure what you mean by unified (do you want all the scenes to have the same resolution?) but you'd need to implement that yourself. You can use the seg map images to figure out the rooms' sizes however.

closed 2022-01-06 by cgokmen

sycz00, 2022-01-08:
Yea I expected that :/ but thanks anyway for your answer :)