StanfordVL/iGibson #150: Issue in renderer

sycz00, 2021-12-24:
Dear iGibson Team,

I found out that there must be a bug somewhere in the renderer. I am using iGibson 2.0.3.
The issue appears when I want to "magically" open one or more objects to see whats inside.

Thats the code for importing objects
`cabinet_dir = os.path.join(igp, "objects/bottom_cabinet/cabinet_0013")
        cabinet_filename = os.path.join(cabinet_dir, "cabinet_0013.urdf")
        
        self.cabin1 = URDFObject(filename=cabinet_filename, name='cabinet1', category="bottom_cabinet")#, scale=np.array([1.0, 1.0, 1.0]), merge_fixed_links=False)
        env.simulator.import_object(self.cabin1,class_id=90)`

To spawn the object I use `env.land(self.cabin1, pos, orn)`

and now to open the object I use the 
`env.scene.open_all_objs_by_category(category="bottom_cabinet",mode="max")`
and to close the object 
`env.scene.open_all_objs_by_category(category="bottom_cabinet",mode="zero")`
respectively. 

Now whats happening is that, in the gui interface, the object opens and closes correctly but the robot camera can't see any changes. Therefore, I assume that the camera render process is not updating the changes. I took some time to see if anything obvious in the simulator and renderer file is wrong but it seems all normal. 

My first assumption was that the object does not get syncronized. Therefore, I used simulator.sync() and did manual pybullet steps but that did not solve the problem.

To further test if this bug comes from the fact that I import objects, I used the already in the scene existing fridge (Rs_Int). The same issue appears to the fridge. 

Unexpectedly, in some of the cases when the robot is very close to the changing object, it is displayed in the robot camera correctly. But this observation is not consistent such that it is not happening all the time.

ChengshuLi, 2021-12-24:
Could you please send us a minimal repreduction script to trigger this issue. Thank you.

sycz00, 2021-12-27:
Well I am using it in a big project. I tried to put it into a small script but without robot manual navigation. So the idea here is that the fridge in the Scene Rs_int opens visually displayed in gui mode but is not rendered in robot frame. Sorry for the bad code embedding (I had not enough time and wanted to finally post it)

`s = Simulator(mode="gui", image_width=128, image_height=128, device_idx=0)
    scene = InteractiveIndoorScene("Rs_int", texture_randomization=False, object_randomization=False)#EmptyScene()
    s.import_ig_scene(scene)
    

    config = parse_config(os.path.join(igibson.example_config_path, "turtlebot_point_nav.yaml"))
    turtlebot = Turtlebot(config)
    s.import_robot(turtlebot)

    try:
        while True:
            
            ac = input()
            
            
            if(ac == "o"):
                scene.open_all_objs_by_category(category="fridge",mode="max")
                action = np.array([0.0,0.0])
            else:
                scene.open_all_objs_by_category(category="fridge",mode="zero")
                action = np.array([0.0,0.0])
           
                     
            action = np.random.uniform(-1, 1, turtlebot.action_dim)
            turtlebot.apply_action(action)
            s.step()
    finally:
        s.disconnect()`

fxia22, 2021-12-27:
@sycz00 This issue is caused by the fact that the object is in sleep. Your solution works, by force waking up the object.

Alternatively, you can call `simulator.sync(force_sync=True)`, this will sync up the pose of objects regardless of their sleep states. 

We let objects go into sleep to accelerate the simulation, if you use physical simulation instead of magic actions, this problem won't happen. We will think about how to better support magic actions with our sleeping mechanism, thanks for your feedback.

cgokmen, 2022-01-06:
Closing this unless there are further issues.

closed 2022-01-06 by cgokmen