StanfordVL/iGibson #151: How to run iGibson in headless mode?

kamalesh0406, 2021-12-24:
I installed iGibson on a remote desktop using pip install and when I try to run the demo I get the following error. 

```
startThreads creating 1 threads.
starting thread 0
started thread 0
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
No protocol specified

	cannot connect to X server
```

I believe this error is because it is trying to render a GUI output. How should I modify the code to run it in headless mode? Thank you!

fxia22, 2022-01-06:
It looks like you try to start the pybullet gui in headless mode, which is not supported. You can set the mode to "headless" and it will run iG in headless mode. 

https://github.com/StanfordVL/iGibson/blob/c0dba21952eb8a9323278be276ea9295425a1f7c/igibson/envs/igibson_env.py#L40

mjlbach, 2022-01-06:
On the latest release, you must set use_pb_gui = True in order to use the pb_gui which is off by default, so this shouldn't happen anymore.

closed 2022-01-06 by mjlbach