StanfordVL/iGibson #152: No module named 'bddl.activity'

WangJuan6, 2022-01-07:
Dear iGibson Team,
    I have installed the igibson sim, bddl, and behavior by the instructions provided at https://docs.google.com/document/d/1u2m9Ld6Qo3eG-fvCzuAZN6lHxwpVVBlwLVdzo_WDlNI. 
  However, when I'm running: python -m igibson.examples.environments.env_nonint_example, there is an erro: 

> Traceback (most recent call last):
>   File "/home/wj/anaconda3/envs/igibosn/lib/python3.8/runpy.py", line 194, in _run_module_as_main
>     return _run_code(code, main_globals, None,
>   File "/home/wj/anaconda3/envs/igibosn/lib/python3.8/runpy.py", line 87, in _run_code
>     exec(code, run_globals)
>   File "/home/wj/projects/iGibson/igibson/examples/environments/env_nonint_example.py", line 8, in <module>
>     from igibson.envs.igibson_env import iGibsonEnv
>   File "/home/wj/projects/iGibson/igibson/envs/igibson_env.py", line 19, in <module>
>     from igibson.tasks.behavior_task import BehaviorTask
>   File "/home/wj/projects/iGibson/igibson/tasks/behavior_task.py", line 7, in <module>
>     from bddl.activity import (
> ModuleNotFoundError: No module named 'bddl.activity'

Thanks,
Juan

mjlbach, 2022-01-07:
Should be fixed by https://github.com/StanfordVL/bddl/releases/tag/0.0.5, currently uploading to pypi :)

closed 2022-01-07 by mjlbach

WangJuan6, 2022-01-08:
Thanks！It worked！