StanfordVL/iGibson #153: RuntimeError: Caught an unknown exception!

WangJuan6, 2022-01-08:
Dear iGibson Team,
I run: python -m igibson.examples.environments.env_nonint_selector, and select the scene of Bowmore, there is an error:

```
Traceback (most recent call last):
  File "/home/wj/projects/iGibson/env_nonint_selector.py", line 44, in <module>
    main()
  File "/home/wj/projects/iGibson/env_nonint_selector.py", line 32, in main
    env.reset()
  File "/home/wj/projects/iGibson/igibson/envs/igibson_env.py", line 479, in reset
    self.task.reset(self)
  File "/home/wj/projects/iGibson/igibson/tasks/task_base.py", line 47, in reset
    self.reset_agent(env)
  File "/home/wj/projects/iGibson/igibson/tasks/point_nav_random_task.py", line 72, in reset_agent
    reset_success = env.test_valid_position(
  File "/home/wj/projects/iGibson/igibson/envs/igibson_env.py", line 413, in test_valid_position
    has_collision = self.check_collision(body_id)
  File "/home/wj/projects/iGibson/igibson/envs/igibson_env.py", line 361, in check_collision
    self.simulator_step()
  File "/home/wj/projects/iGibson/igibson/envs/env_base.py", line 230, in simulator_step
    self.simulator.step()
  File "/home/wj/projects/iGibson/igibson/simulator.py", line 459, in step
    self.sync()
  File "/home/wj/projects/iGibson/igibson/simulator.py", line 473, in sync
    self.viewer.update()
  File "/home/wj/projects/iGibson/igibson/render/viewer.py", line 75, in update
    frames = self.renderer.render_robot_cameras(modes=("rgb"))
  File "/home/wj/projects/iGibson/igibson/render/mesh_renderer/mesh_renderer_cpu.py", line 1225, in render_robot_cameras
    for item in self.render(modes=modes, hidden=hidden_instances):
  File "/home/wj/projects/iGibson/igibson/render/mesh_renderer/mesh_renderer_cpu.py", line 883, in render
    self.optimize_vertex_and_texture()
  File "/home/wj/projects/iGibson/igibson/render/mesh_renderer/mesh_renderer_cpu.py", line 1292, in optimize_vertex_and_texture
    self.tex_id_1, self.tex_id_2, self.tex_id_layer_mapping = self.r.generateArrayTextures(
RuntimeError: Caught an unknown exception!
```

Thanks,
Juan

roberto-martinmartin, 2022-01-10:
Hi @WangJuan6 ,
we are looking into your issue. Could you let us know what is your system configuration (machine, OS...). 
Also, could you trying scaling the texture to see if it is an issue of VRAM? There are several examples that show how to scale texture by 0.5
Thanks!

WangJuan6, 2022-01-12:
Dear iGibson Team,
Thanks for your reply.

My system configuration is as follows:
- Ubuntu 20.04
- NVIDIA RTX3070
- Nvidia driver: 470.86
- CUDA: 11.1
- CuDNN: 8.0.5
- CMake: 3.22.0
- g++: 9.3.0

I added config_data["texture_scale"] = 0.5 to env_nonint_selector.py and set texture_scale: 0.5 in turtlebot_static_nav.yaml. However, the program still has the same error.

Thanks,
Juan

roberto-martinmartin, 2022-01-12:
Thanks!
Does the error appear with other scenes or only with Bowmore? Could you try "Rs", our demo environment?

WangJuan6, 2022-01-12:
Dear iGibson Team,

I run: python -m igibson.examples.environments.env_nonint_selector, and test 572 scene of gibson_v2_all.tar.gz, I found error appear in these scenes:

WangJuan6, 2022-01-12:
Dear iGibson Team,
I run: python -m igibson.examples.environments.env_nonint_selector, and

WangJuan6, 2022-01-12:
Dear iGibson Team,

I run: python -m igibson.examples.environments.env_nonint_selector, and test 572 scenes of gibson_v2_all.tar.gz, the error appear in these scenes:

`['Alfred', 'Apache', 'Bethlehem', 'Biltmore', 'Bonnie', 'Bowlus', 'Bowmore', 'Brown', 'Bushong', 'Cayuse', 'Channel', 'Chesaning', 'Cottonport', 'Crugers', 'Deatsville', 'Duluth', 'Eagan', 'Frankton', 'Gough', 'Grace', 'Gracey', 'Grainola', 'Grassy', 'Gratz', 'Gravelly', 'Hindsboro', 'Hurley', 'Keweenaw', 'Kinney', 'Kirwin', 'Klickitat', 'Lacon', 'Leilani', 'Lessley', 'Lincolnwood', 'Lindenwood', 'Lucan', 'Maguayo', 'Marstons', 'Micanopy', 'Mifflintown', 'Moark', 'Mogadore', 'Moonachie', 'Nemacolin', 'Ooltewah', 'Pearce', 'Portal', 'Poyen', 'Purple', 'Ruckersville', 'Sargents', 'Scioto', 'Seatonville', 'Shelbyville', 'Sherrill', 'Stigler', 'Terrell', 'Thrall', 'Tillman', 'Tysons', 'Upham', 'Waimea', 'Wappingers', 'Weleetka', 'Wilkinsburg', 'Willow', 'Winooski']`

Thanks,
Juan

ChengshuLi, 2022-01-12:
@WangJuan6 Thank you very much for your thorough test. This is very helpful.
@fxia22 Do you know what might be the reason? Are those texture files corrupted?