StanfordVL/iGibson #154: Efficient way of training agent on multiple env's

sycz00, 2022-01-08:
Hey,

Just a quick question: is there a way to increase the speed of using different scenes ? I found the function 'reload_model' in iGibson_env.py but that consumes a lot of computations when changing the scene after each reset. 
Furthermore, some sort of issue came up by changing the scene in "gui" mode of the simulator (after changing the scene three times). In my case the program just crashes.
Here just a snapshot of the stacktrace:

`b3Printf: Selected demo: Physics Server `
`startThreads creating 1 threads.`
`starting thread 0`
`started thread 0 `
`MotionThreadFunc thread started`
`Segmentation fault (core dumped)`

When using the "headless" mode, the issue doesn't pop up.

mjlbach, 2022-02-04:
The best way to train on multiple scenes is to use multiple environment workers, here is a snippet from one of my projects showing how to leverage 'scene_id' and 'worker_index' (from ray) to do this.

```
def igibson_env_creator(env_config):
    from igibson.envs.igibson_env import iGibsonEnv

    # Parllelize environments across workers
    env_config['scene_id'] = env_config['scene_ids'][env_config['worker_index']]

    return iGibsonEnv(
        config_file=env_config,
        mode=env_config["mode"],
        action_timestep=1 / 10.0,
        physics_timestep=1 / 40.0,
    )

@hydra.main(config_path=ssg.CONFIG_PATH, config_name="config")
def main(cfg):
    ray.init(local_mode=cfg.debug)
    env_config = OmegaConf.to_object(cfg)
    register_env("igibson_env_creator", igibson_env_creator)
    checkpoint_path = Path(cfg.experiment_save_path, cfg.experiment_name)

    if cfg.scene_ids:
        assert len(cfg.scene_ids) == cfg.num_workers, "Error, number of workers must match number of scenes in config file"

    config = {
        "env": "igibson_env_creator",
        "model": {
            "conv_filters": [[32, 8, 4], [64, 4, 2], [64, 3, 1]],
        },
        "env_config": env_config,  # config to pass to env class
        "num_workers": cfg.num_envs,
        "framework": "torch",
        "evaluation_interval": 100, # every n episodes evaluation episode
        "evaluation_config": {
            "callbacks": DummyCallback,
            "record_env": True,
            # "render_env": True,
        },
        "lambda": cfg.gae_lambda,
        "lr": cfg.learning_rate,
        "train_batch_size": cfg.n_steps,
        "num_sgd_iter": cfg.n_epochs,
        "sgd_minibatch_size": cfg.batch_size,
        "gamma": cfg.gamma,
        "num_gpus": 1,
        "callbacks": MetricsCallback,
    }

    log_path = str(checkpoint_path.joinpath("log"))
    trainer = ppo.PPOTrainer(
        config,
        logger_creator=lambda x: UnifiedLogger(x, log_path),
    )

    if Path(checkpoint_path).exists():
        checkpoints = Path(checkpoint_path).rglob("checkpoint-*")
        checkpoints = [
            str(f) for f in checkpoints if re.search(r".*checkpoint-\d*$", str(f))
        ]
        checkpoints = sorted(checkpoints)
        if len(checkpoints) > 0:
            trainer.restore(checkpoints[-1])

    for i in range(cfg.training_timesteps // cfg.n_steps):
        # Perform one iteration of training the policy with PPO
        trainer.train()

        if i % 5 == 0:
            checkpoint = trainer.save(checkpoint_path)
            print("checkpoint saved at", checkpoint)

if __name__ == "__main__":
    main()
```

sycz00, 2022-02-07:
Hey,

thanks for your answer. Okay so what you essentially do is, you set the scene for each of your parallel env's individually, e.g. 5 workers with 5 scenes.
This might differ from my case in which I am only able to train on 8 workers with ~10 - 12 scenes and therefore I must rely on changing the scene during the run.

closed 2022-02-20 by sycz00