StanfordVL/iGibson #156: Inverse Kinematics example is not up-to-date

CharlesAverill, 2022-01-20:
[The Inverse Kinematics example script](https://github.com/StanfordVL/iGibson/blob/master/igibson/examples/robots/ik_example.py) does not work out-of-the-box, and will error out with a message about `control_freq` being specified in Fetch's configuration file. When this error is bypassed by commenting out the assertion, the resulting simulation flails Fetch's arm wildly.

This script is fairly important for outlining the usage of IK in iGibson. If I fix it, I will submit a PR. Just wanted to outline the issue here as well.

CharlesAverill, 2022-01-20:
Turns out the initial assertion issue is because `fetch = Fetch(config)` should be something like `fetch = Fetch(**config["robot"])`, although this still causes issues as the configuration contains values that the constructor does not accept, like `name`. This can be remedied by adding `**kwargs` as the final parameter in Fetch's constructor, although this does not solve the flailing issue.

ChengshuLi, 2022-01-20:
@CharlesAverill Thank you for reporting this. We are aware that several of our examples are currently broken (including the IK example). We are in the process of fixing them (and making them much better than before!). We will release a new version in a week or two.

CharlesAverill, 2022-01-20:
@ChengshuLi Got it, thank you! Loving the library so far, can't wait to see where it goes.

CharlesAverill, 2022-02-12:
Hi @ChengshuLi, is the next version release still planned to be soon?

cgokmen, 2022-02-12:
We are at code-complete and are planning to release in the upcoming days after completing testing.

cgokmen, 2022-02-17:
We have pushed the release. Let us know if you need any additional help!

closed 2022-02-17 by cgokmen

CharlesAverill, 2022-02-25:
@cgokmen IK runs without a hitch now. One thing I'm curious about: does iGibson support any sort of motion planning to move an appendage from its current position to another position (computed by IK in my case)? I'm currently working on porting some motion planning code, but it's proving to be tough, an official solution would be much preferred.

cgokmen, 2022-02-25:
We do have some motion planning code available (see MotionPlanningWrapper in utils/ and the stuff in external/ it calls). Unfortunately currently it gives you the plan but no easy way to execute it. We are planning to release the functionality to execute these in our next (and final) release of iGibson v2, but I'm hesitant to give a timeline currently.

CharlesAverill, 2022-03-01:
@cgokmen The motion planning example is helpful, but one thing that's a bit frustrating is the [hardcoded gripper position + orientation](https://github.com/StanfordVL/iGibson/blob/master/igibson/examples/robots/motion_planning_example.py#L137). 

[iGibson's official IK implementation](https://github.com/StanfordVL/iGibson/blob/master/igibson/examples/robots/ik_example.py#L136) keeps physics active during calculation, causing the arm to send objects it collides with flying. I've also looked at [pybullet's implementation](https://github.com/StanfordVL/iGibson/blob/783a38ef4962a8757a75c1ece76450163e2c7654/igibson/external/pybullet_tools/utils.py#L3995) but it requires a gripper orientation, which is as big a problem as IK. 

TLDR; the inverse kinematics calculations interact with the environment, making the motion planning calculations inaccurate. Is there an official resolution to this issue?

roberto-martinmartin, 2022-03-01:
Hi @CharlesAverill 
I hope I understand your issues
- With respect to the hardcoded gripper position+orientation, the example just shows how to use IK but the example can be modified for your purpose or you could create your own application based on the example. Maybe I'm not understanding the problem here.
- About the IK implementation, yes, it may collide. However, we save and restore the pybullet state before/after the computation, so the objects should return to their original state. IK only tries to find a joint configuration, not checking for collisions, iirc
- "making the motion planning calculations inaccurate" -> what is the exact issue you found? could you describe it
thanks!

CharlesAverill, 2022-03-01:
Hi @roberto-martinmartin I think I poorly explained how we're utilizing IK and motion planning and the issues we're experiencing. 

The end goal is to generate a set of random objects on a tabletop and push one of them. We envision the solution to be defined as:
- Getting the position of the arm to be at in order to push the object using IK
- Creating a motion plan to move the arm to the position returned by IK
- Moving the arm through that plan, and pushing the object

The issues we're having occur because IK bumps into objects while it's calculating. This causes the motion plan to generate a plan to an arm position where the gripper is no longer near the object it was attempting to reach. I see the calls to pybullet's state save and restore functions in the example, and they are included in our code, but we still get bumping issues. Perhaps it's an implementation issue on our end (likely), I'll try making sure I'm not messing with the environment after it loads the saved state.