StanfordVL/iGibson #158: Place YCBObjects on surface below point

CharlesAverill, 2022-01-24:
I'd like to place a table on an object, and a few objects on said table, programmatically. 

Right now I'm dropping the objects onto the table, and the table onto the floor, because if their height coordinate is too low, they'll clip through the floor and go flying, same with the objects in relation to the table. The downside of dropping is that some objects get knocked over, and I have to run the simulation for some arbitrary amount of time until the objects settle down before I can start recording data, for consistency.

I'm planning to raycast below each object to find the exact height to spawn them at, but I was wondering if iGibson supports or plans to support a method to do this without the user having to delve into pybullet.

Thank you,

Charles

cgokmen, 2022-01-24:
Hi,

Does `object_to_drop.states[object_states.OnTop].set_value(object_with_surface, True)` do what you need?

You can take a look at the example here: https://github.com/StanfordVL/iGibson/blob/master/igibson/examples/objects/sample_objects_logic.py

Let us know if that works!

cgokmen, 2022-01-24:
(note that in order to do this for the floor, you'd use the same line with the `OnFloor` state and a `RoomFloor` object.

CharlesAverill, 2022-01-24:
@cgokmen Would that work with `body_id`s as well? The EmptyScene doesn't keep a reference to the Floor object, just its ID

CharlesAverill, 2022-01-24:
@cgokmen Is there a function to load a RoomFloor object from an already-instantiated `body_id`? `EmptyScene` does not keep a reference to the floor it spawns

cgokmen, 2022-01-24:
It doesn't work with body IDs, but the on-floor sampling is already fairly trivial. It looks something like:

```
pos = # You decide how you're going to sample a 3D position on the floor (e.g. at the floor height)

# Get the AABB of the object.
lower, _ = objA.states[object_states.AABB].get_value()
# Move the position to a stable Z for the object.
pos[2] += objA.get_position()[2] - lower[2]
```

And then some other checks are done to drop the object & make sure it's not in collision. See the full implementation in https://github.com/StanfordVL/iGibson/blob/d416a470240eb7ad86e04fee475ae4bd67263a7c/igibson/object_states/utils.py#L80

I'll add it to our roadmap to make EmptyScene generate a floor object too, thanks!

CharlesAverill, 2022-01-24:
Floor placing looks good, although I'm not noticing a difference when placing down YCBObjects. Gonna switch them out for URDFs since those seem to work fine. Thank you for your help!

closed 2022-01-24 by CharlesAverill