StanfordVL/iGibson #160: Using CVPR2021 Challenge Code: Decryption Errors

felipefelixarias, 2022-01-25:
I intend to use the social navigation features from the CVPR2021 challenge for my research and am having issues running the code given some changes that have been made to Gibson since. The objects and meshes are now encrypted and I am using the challenge branch so I get the errors below for a variety of objects:

/.../gibson2/data/ig_dataset/scene_instances/20220125-125018_3046166092929093048_30752/: cannot extract anything useful from mesh '/.../gibson2/data/ig_dataset/objects/window/103238/shape/visual/link_2_m1_vm.encrypted.obj' b3Warning[examples/SharedMemory/plugins/tinyRendererPlugin/TinyRendererVisualShapeConverter.cpp,558]: issue extracting mesh from COLLADA/STL file /.../gibson2/data/ig_dataset/objects/window/103238/shape/visual/link_2_m1_vm.encrypted.obj

I've added the key to the data folder as per the new instructions, but I would need to add the decryption code from master. Have these changes for compatibility been implemented anywhere? or are there any suggested fixes? This issue is related to #159 #126 and #119, but I have different constraints given that I can't use master/latest code or the docker images.

fxia22, 2022-01-26:
You might be using a wrong version of ig_dataset, for CVPR21 challenge, the ig_dataset is not encrypted, so decoding is not needed. You should download the dataset from this link https://storage.googleapis.com/gibsonchallenge/gibson_challenge_data_2021.tar.gz, or use this script: https://github.com/StanfordVL/iGibsonChallenge2021/blob/master/download.sh

jing-bi, 2022-01-26:
I'm using the dataset downloaded with `python -m igibson.utils.assets_utils --download_ig_dataset` command and still encounter the same problem as mentioned in #159. The igibson.key is in the igibson/data folder.

Not sure if this is the problem with the key or with the asset?

felipefelixarias, 2022-01-28:
Thank you @fxia22! I will use the challenge version of the dataset instead. Do you know if social navigation will be supported in master in the future?

felipefelixarias, 2022-02-07:
@fxia22 One more question, were the dev and/or test environments made available after the end of the challenge? I can't use the same code to create a dev and test set using the iGibson 2 environments (because they are encrypted). Should I just use gibson 1 environments at the cost of interactive objects?

closed 2022-06-03 by felipefelixarias