StanfordVL/iGibson #161: error related to `verify_determinism()` function

jing-bi, 2022-01-25:
i'm trying to replay the raw vr recording with `safe_replay_demo`
despite the bounch of warning for loading material
```shell
Warning[examples/SharedMemory/plugins/tinyRendererPlugin/TinyRendererVisualShapeConverter.cpp,558]:
issue extracting mesh from COLLADA/STL file
```

I found it's always impossiable to pass the `verify_determinism()` function, the object list never match in `original_file["physics_data"]` and `new_file["physics_data"]`.

For instance, the replay of `cleaning_windows_0_Rs_int_0_2021-05-23_23-11-46.hdf5` will yield two different set of objects:
```python
new_file['physics_data'].keys()
Out[3]: <KeysViewHDF5 ['1', '100', '119', '120', '121', '122', '123', '194', '58', '59', '8', '90']>
original_file["physics_data"].keys()
Out[4]: <KeysViewHDF5 ['1', '194', '45', '46', '59', '60', '62', '63', '64', '65', '66', '8']>
```

Not sure if they are related, please let me know if you need more info.

jing-bi, 2022-01-26:
After a few hours of debugging, seems the behavior robot location is different from the one in the behavior dataset and cannot adjust its position when executing the actions.
For example, the camera will bump into the ceiling in the replayed video.

cgokmen, 2022-01-26:
Is this with the `behavior-replay` branch of iGibson (which you can get from the GitHub repo)? You also need to be on the behavior-replay version of bddl and the dataset.

jing-bi, 2022-01-26:
I'm trying right now

jing-bi, 2022-01-26:
behavior-replay branch of iGibson + bddl 0.0.5 commit 
After a few modifications, I finally make it able to run, but still cannot replay the raw VR data.
```
Demo was succesfully completed:  False
IG LOGGER INFO: Ending log writing session after 3000 frames
Mismatch for obj 194 with mismatched attribute orientation starting at timestep 982
Mismatch for obj 194 with mismatched attribute position starting at timestep 983
Mismatch for obj 45 with mismatched attribute joint_state starting at timestep 0
Mismatch for obj 59 with mismatched attribute joint_state starting at timestep 1905
Mismatch for obj 60 with mismatched attribute joint_state starting at timestep 2614
Mismatch for obj 62 with mismatched attribute orientation starting at timestep 612
Mismatch for obj 62 with mismatched attribute position starting at timestep 913
Mismatch for obj 63 with mismatched attribute orientation starting at timestep 0
Mismatch for obj 63 with mismatched attribute position starting at timestep 834
Mismatch for obj 64 with mismatched attribute orientation starting at timestep 0
Mismatch for obj 64 with mismatched attribute position starting at timestep 19
Mismatch for obj 65 with mismatched attribute orientation starting at timestep 0
Mismatch for obj 65 with mismatched attribute position starting at timestep 11
Mismatch for obj 66 with mismatched attribute orientation starting at timestep 0
Mismatch for obj 66 with mismatched attribute position starting at timestep 12
```

I'm worrying if the new change in master branch will affect the evaluation of the Behavior benchmark result.

roberto-martinmartin, 2022-01-26:
yes, master branch has diverged from what was used at the time of collecting the demos. that's why the corresponding branches should be used to work with the VR data. Those branches have been kept as they were for back compatibility with the VR demos and should work

roberto-martinmartin, 2022-01-26:
what commands are you concretely running?

roberto-martinmartin, 2022-01-26:
also, could you point me to the instructions you are following in your development? I'd like to see if the instructions are correct, up to date and complete. Thanks

jing-bi, 2022-01-26:
The instruction I'm following is from here: https://stanfordvl.github.io/behavior/installation.html
By default, the repo I downloaded should all be the latest master commit, which might cause some problems.

Say what if we want to evaluate the method, do we have to go back to the deprecated version to do so? I asked because after i think the development of a newer environment is more concise and clean at least easier to understand for me. is there a way we can use the newer environment to do the replay?

jing-bi, 2022-01-26:
btw, at commit fe9e80b, the behavior replay is compatible with the refactored environment, but I'm still struggling to replicate the result.

jing-bi, 2022-01-26:
I generated a video so maybe it's easier to see what's wrong

https://user-images.githubusercontent.com/34174970/151198887-15319586-04fe-47d8-aae3-1d592f7d0c66.mp4