StanfordVL/iGibson #162: PointNav Task

asfandasfo, 2022-02-02:
Hi, I was trying to train pointnav agent using the given example 'stable_baselines3_example.py' but It gives me a memory error(attached). I solve this by reducing 'num_environments' from 8 to 1. But it isn't converging. I also attached the tensorboard [logs.](https://tensorboard.dev/experiment/N0ArHRnWSJ2eo3JY5fdegA/#scalars) Do I need to change any other parameters (e.g. learning rate etc) to make it work with 1 environment.?
[image]

sycz00, 2022-02-07:
Well I let it run with 5 or 6 parallel env's and it kinda converged too. Therefore, I am not sure if it converges with one. However, multiple environments are particularly important for on-policy algorithms like PPO because it reduces the overall correlation of your samples. In this specific example I would say that you must adapt the lr and the number of updates per PPO training due to the reduced rollout-buffer size.

asfandasfo, 2022-02-07:
> Well I let it run with 5 or 6 parallel env's and it kinda converged too. Therefore, I am not sure if it converges with one. However, multiple environments are particularly important for on-policy algorithms like PPO because it reduces the overall correlation of your samples. In this specific example I would say that you must adapt the lr and the number of updates per PPO training due to the reduced rollout-buffer size (no guarantees)

I tried using 8 environments but with a smaller input image(140x160) and it's still not converged.

sycz00, 2022-02-07:
According to you logs, the training seems to be quite unstable. As far as I remember, the success rate for this task was at roughly 50 - 60%. But in your case something different is wrong. Have you checked your env in the sense that every sample for you agent is correct ? Again accordingly to you logs, PPO might need completely different HP's (try reducing lr with constant magnitude, reduce n_steps, set clip_parameter to 0.1). Furthermore, observe what you explained variance is doing, it should "generally", gradually increase during training. In case nothing of this works, execute the tests for iGibson and run unit tests for SB3 as well.

asfandasfo, 2022-02-18:
> According to you logs, the training seems to be quite unstable. As far as I remember, the success rate for this task was at roughly 50 - 60%. But in your case something different is wrong. Have you [...]

Is there any pre-trained model available for pointnav task?

sycz00, 2022-02-20:
not as far as I know.

genkv, 2022-03-23:
@asfandasfo I encountered the exact same issue with you. Have you solved this problem? thanks.

asfandasfo, 2022-03-24:
> @asfandasfo I encountered the exact same issue with you. Have you solved this problem? thanks.

If you decrease the learning rate, it does converge but validation performance is still poor.

genkv, 2022-03-24:
> According to you logs, the training seems to be quite unstable. As far as I remember, the success rate for this task was at roughly 50 - 60%. But in your case something different is wrong. Have you [...]

@sycz00 I had the same problem as OP. I followed your suggestions and here's the [result](https://tensorboard.dev/experiment/laKd7CV5TkqoJqS55ryDZg/#scalars). Still very unstable.

I began to realize that the memory needed for the rollout buffer in stable_baselines3.py is huge. At least (task_obs + rgb image + depth + scan) * 8 <num_environments> * 2048 <batch_size>. To make it get the program started, we need to reduce observation space dimensions. I took out the depth and scan, thus only using task_obs and rgb in observation space.

I wonder what a "decent" or "normal" training process look like for any PointNav task. The epoch reward mean should gradually increase, right?

sycz00, 2022-03-27:
@genkv 
Hey, 

I think if you leave out depth images, the example could potentially never work.
According to your logs, two things were standing out. Firstly, the learning rate (which doesn't look that big), seems to cause big updates according to the KL-div plot. This could explain why the mean reward is alternating and the agent never learns properly.
Another thing that I saw was the speed of your std. decrease. Is the entropy loss coefficient too big ? This graph should definitely be decreasing but ending at 0.1 looks like the policy gets too deterministic.
The other plots do not look that bad.
And yes, the mean reward should increase. 
Another hint: when evaluating the checkpoint in SB3, try using the stochastic policy instead of the deterministic policy. Because only for the stochastic policy is ensured to be optimized as in the training process.

mjlbach, 2022-04-13:
Here's a simplified example of object nav that will train (slightly harder than point nav), see this comment:

https://github.com/StanfordVL/iGibson/issues/180#issuecomment-1097453051