StanfordVL/iGibson #164: Matterport dataset with semantic mask

sycz00, 2022-02-20:
Hey,

I trying to evaluate my RL model and would need to do so on the Matterport dataset. 
This is probably an rhetorical questions but I wanted to know if there's any chance to get some semantic segmentation mask on non-iGibson datasets such as matterport ?

Thanks for your help.

cgokmen, 2022-02-25:
Matterport does provide instance segmentation data, but as far as I know we do not import this data into iGibson. Theoretically the same system that does the iG scene semantic mask could be used to do this, you'd only need to figure out how to import the data. @fxia22 do you have any additional info about this?

sycz00, 2022-03-02:
Hey thanks for your answer.  
I will try it soon. 
I have another quick questions which you might know. I wanted to know the sizes of the iGibson scenes in square meters but could not find any official label for it.

roberto-martinmartin, 2022-03-22:
I believe the information you need is in the first paper/version of the simulator (https://github.com/StanfordVL/GibsonEnv/blob/master/gibson/data/README.md#navigation-benchmark-scenarios).

See the "metadata"

closed 2022-03-22 by roberto-martinmartin