StanfordVL/iGibson #165: Dynamic robot and object spawning

CharlesAverill, 2022-02-24:
Is there an official way to dynamically spawn a robot into an iGibson house scene? Our team has a jury-rigged process we've been using involving pathfinding between random points sampled from the floor, but I would imagine there is an officially-supported way to accomplish this.

sycz00, 2022-02-24:
Perhaps you could take a look at` env.test_valid_position(object, pos, orn)` to verify that the object gets there without collision and `env.land(obj_list[0], pos, orn)` for landing the object. Furthermore, use 
`_, initial_pos = env.scene.get_random_point(floor=self.floor_num)` to sample a new position (robot or object) and use 
`reset_success = env.test_valid_position(env.robots[0], initial_pos, initial_orn)` to check if the spawn point for the agent is a proper choice.

cgokmen, 2022-02-25:
What are the requirements re: the sampled position? The method recommended above by @sycz00 sounds reasonable.

CharlesAverill, 2022-02-25:
@sycz00 Thank you for the tips, we'll be trying this very soon! @cgokmen I agree that the solution good. As for requirements, we would just like to find a position with enough space to spawn a table and robot, consistently.

roberto-martinmartin, 2022-03-22:
Closing this issue. Feel free to reopen

closed 2022-03-22 by roberto-martinmartin