StanfordVL/iGibson #166: How to safely remove doors - without crashing the step-function?

meier-johannes94, 2022-02-25:
I modified the Behavior Challenge in the discrete case a little bit, where I also reposition the objects. Now the problem are the doors. Because they seem closed and in the discrete case we can't open them. 

**What I already tried:** 
I removed the doors via PyBullet's p.removeBody(id) functionality. However then I am getting this error, when I call the step function of the environment: 
```
  File "/is/sg2/jmeier/behaviorchallenge/iGibson/igibson/envs/behavior_env.py", line 262, in step
    self.simulator.step()
  File "/is/sg2/jmeier/behaviorchallenge/iGibson/igibson/simulator.py", line 1140, in step
    self.sync()
  File "/is/sg2/jmeier/behaviorchallenge/iGibson/igibson/simulator.py", line 1150, in sync
    self.body_links_awake += self.update_position(instance, force_sync=force_sync)
  File "/is/sg2/jmeier/behaviorchallenge/iGibson/igibson/simulator.py", line 1803, in update_position
    dynamics_info = p.getDynamicsInfo(instance.pybullet_uuid, link_id)
```

Would you please tell me how to remove objects and still being able to call the step() function? 
If there is no way this works or if you have additionally other suggestions I am quite open. 
Thank you very much in advance!

cgokmen, 2022-02-25:
Try passing the `not_load_object_categories=["door"]` argument to the scene init function. A simple way of doing this is adding it into the config.yml file for your environment:

```
not_load_object_categories:
  - door
```

meier-johannes94, 2022-02-25:
@cgokmen 
Thanks! That worked perfectly! Great!

closed 2022-02-25 by meier-johannes94