StanfordVL/iGibson #168: Can I stop the prints, when doing an environment reset?

meier-johannes94, 2022-03-02:
When I do a reset, then there is a lot of information written into the console about what module is loaded, etc. 
Is it possible to deactivate this completely? 
This would be quite useful for debugging.

roberto-martinmartin, 2022-03-02:
Hi @meier-johannes94,
are you using the latest version? we cleaned up a lot of the printing, even the pybullet prints, so the command line should be very clean.
We also use more strictly the `logging` package so if you set the logging level higher, you should see even less messages.

meier-johannes94, 2022-03-02:
@roberto-martinmartin 
Thanks for the quick answer! I am not using the most recent one. My project has a close deadline, so I would prefer not to switch now. Is there also an option in the old version that was available around December?

roberto-martinmartin, 2022-03-02:
I see. You could still try to set the logging level higher but many prints were not using logging, and they won't be affected (that is what we fixed in the last version). I hope that helps a bit!
A lot of the printing when loading models comes from pybullet, though, and that won't go away. Maybe you could try to get the latest version of the pybullet-svl package

closed 2022-03-22 by roberto-martinmartin