StanfordVL/iGibson #169: Low FPS in Non-Headless Mode

GaganBhat, 2022-03-04:
During headless mode in a demo scene, I can see around ~70-80 FPS. 

However, when the visualization is done the FPS drops to an incredibly low 5-7 FPS. 

This is with an RTX A5000 and Intel i9 machine. Is this level of performance expected?

cgokmen, 2022-03-04:
Hi, what are you trying to use iGibson for? The particular demo you are running uses a Gibson scene and not an iGibson one - these scenes are from the first Gibson paper where non-interactive scenes were built out of 3d scans of real environments. I wonder if that is the reason - are you intentionally trying one of these scenes, or are you trying to use our interactive scenes (from the iGibson papers)?

penttinj, 2022-03-09:
Hi, I've also noticed that version 2.0.6 causes a huge performance drop for the non-interactive example. 2.0.5 runs at 21-24 FPS while 2.0.6 runs the same 5-7 FPS as OP.

ChengshuLi, 2022-03-09:
Thank you for reporting the performance regression. We are currently working on it and will get back to you.

GaganBhat, 2022-03-10:
@cgokmen I am using iGibson primarily for exploration algorithms for Fetch in previously unseen environments. In this use case, I'm primarily using the Matterport3D dataset which contains non-interactive 3D scans of real environments.

cgokmen, 2022-03-10:
Thanks for the reports - we have fixed the issue and will be releasing the fix in a few hours.

cgokmen, 2022-03-10:
@GaganBhat and @penttinj thanks again for both of your reports - we have released iGibson 2.1.0 which corrects this issue, feel free to switch to that. Let us know if you still run into issues.

closed 2022-03-10 by cgokmen