StanfordVL/iGibson #171: Failing building wheel for igibson

kevin-thankyou-lin, 2022-03-07:
Hi! Just running into an issue installing igibson via pip and a conda env on Ubuntu 18.04:

```
  creating build/lib.linux-x86_64-3.8/igibson/render/mesh_renderer/shaders/deprecated
...
  running build_ext
  AttributeError: DistutilsMetaFinder instance has no attribute 'find_module'
  Traceback (most recent call last):
    File "/home/thankyou-always/miniconda3/envs/igibson/lib/python3.8/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 349, in <module>
      main()
...
    File "/tmp/pip-build-env-84scm5xk/overlay/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 986, in run_command
      cmd_obj.run()
    File "setup.py", line 44, in run
      out = subprocess.check_output(["cmake", "--version"])
...
    File "/home/thankyou-always/miniconda3/envs/igibson/lib/python3.8/subprocess.py", line 516, in run
      raise CalledProcessError(retcode, process.args,
  subprocess.CalledProcessError: Command '['cmake', '--version']' returned non-zero exit status 1.
  ----------------------------------------
  ERROR: Failed building wheel for igibson
  Building wheel for bddl (setup.py) ... done
  Created wheel for bddl: filename=bddl-1.0.0-py3-none-any.whl size=210133 
  Building wheel for gym (PEP 517) ... done
  Created wheel for gym: filename=gym-0.22.0-py3-none-any.whl size=708394 sha256=b74ac798a03c92c99bdcee8b74949125888275e7a6907b30dd481c02253b48ad
...
  Building wheel for py360convert (setup.py) ... done
  Created wheel for py360convert: filename=py360convert-0.1.0-py3-none-any.whl size=8162 sha256=c6b1aa1faaa3b426cf4629887619a31d4b5d5bffa0dc5773b5c02359392e8609
  Stored in directory: /home/thankyou-always/.cache/pip/wheels/7b/7d/05/75fc72b58f90f3bba677f505ba5c4e7893610216f2aef42960
Successfully built bddl gym progressbar pybullet-svl transforms3d GPUtil py360convert
Failed to build igibson
ERROR: Could not build wheels for igibson which use PEP 517 and cannot be installed directly
```

Would you know what the issue is?

ChengshuLi, 2022-03-07:
Or maybe you don't have `cmake` installed?

kevin-thankyou-lin, 2022-03-07:
Hmm, seems like `DistutilsMetaFinder` not noving `find_module` in that github repo isn't the issue because I installed things successfully on another machine using the exact same commands that failed on my first machine

roberto-martinmartin, 2022-03-22:
I'm going to close this issue since we can't reproduce the problem and @kevin-thankyou-lin was able to install successfully in another machine. We will reopen if it repeats

closed 2022-03-22 by roberto-martinmartin