StanfordVL/iGibson #173: Where can I download the latest ig_dataset

genkv, 2022-03-14:
I followed the instruction from the official documentation and downloaded ig_dataset.tar.gz from https://storage.googleapis.com/gibson_scenes/ig_dataset.tar.gz

After decompressing it and trying to use the data with the latest version of igibson, I got the error complaining that my ig_dataset is outdated as its version is v2.0.1. The latest version of igibson (I have 2.1.0) only accepts ig_dataset whose version is in the range of [2.0.6, 2.1.1)

Where can I download the latest version of ig_dataset, if not from the provided url? Thanks

mjlbach, 2022-03-14:
When I download the latest ig_dataset, the version is:

- wget https://storage.googleapis.com/gibson_scenes/ig_dataset.tar.gz 
- tar xzf ig_dataset.tar.gz
- cat ./ig_dataset/VERSION

It shows 2.0.6, which is the most current version.

genkv, 2022-03-14:
> When I download the latest ig_dataset, the version is:
> 
> * wget https://storage.googleapis.com/gibson_scenes/ig_dataset.tar.gz
> * tar xzf ig_dataset.tar.gz
> * cat ./ig_dataset/VERSION
> 
> It shows 2.0.6, which is the most current version.

the tar.gz file I downloaded has ig_dataset_encrypted/version "2.0.1" and the version file name is lower cased instead of VERSION

genkv, 2022-03-14:
> Did you repeat those steps _exactly_, because it seems like you have an old file.

I did it again and this time I got 2.0.6. So thanks that resolved the issue. I don't know why somehow I managed to download the older version...

closed 2022-03-14 by genkv