StanfordVL/iGibson #174: Installation failed on linux: igibson2

Zishuo-9, 2022-03-16:
Building wheels for collected packages: gibson2
  Building wheel for gibson2 (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [6 lines of output]
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/tmp/cylin/pip-install-bl1i11br/gibson2_6474e5aff0ef4c3bb4f4a2e9f477abdc/setup.py", line 126, in <module>
          raise Exception(message)
      Exception: You tried to install iGibson. The package named for iGibson is "igibson"
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for gibson2
  Running setup.py clean for gibson2
Failed to build gibson2
Installing collected packages: gibson2
  Running setup.py install for gibson2 ... error
  error: subprocess-exited-with-error

  × Running setup.py install for gibson2 did not run successfully.
  │ exit code: 1
  ╰─> [6 lines of output]
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/tmp/cylin/pip-install-bl1i11br/gibson2_6474e5aff0ef4c3bb4f4a2e9f477abdc/setup.py", line 123, in <module>
          raise Exception(message)
      Exception: You tried to install iGibson. The package named for iGibson is "igibson"
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> gibson2

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.

mjlbach, 2022-03-16:
What was the exact pip command you ran? Gibson2 is the package name for an older version of igibson

Zishuo-9, 2022-03-16:
Hi, Thanks! I replace pip install gibson2 with pip install igibson. It is working.

closed 2022-03-16 by mjlbach