StanfordVL/iGibson #175: problem about transformation from image coordinate to world coordinate

WangJuan6, 2022-03-16:
Dear iGibson Team,
I encountered a problem when I want to transform the point cloud from image coordinate to world coordinate.
First, I generate point cloud from depth image;
[image]
Second, I calculate the rotation matrix R using the orientation information from env.simulator.robots[0].get_orientation() and get the current position from env.simulator.robots[0].get_position()
Third, I transform coordinate by:
point_cloud_t_geometric = np.matmul(point_cloud_t.reshape(-1, 3), R.T).reshape(XYZ.shape)
point_cloud_t_geometric[:, :, 0] += current_position[0]
point_cloud_t_geometric[:, :, 1] += current_position[1]
point_cloud_t_geometric[:, :, 2] += current_position[2]
The results are as follows, red means the result at time t, and blue means the result at time t+1.
[image]
Obviously, there is an offset in the point cloud at times t and t+1.
How can I solve this problem?
Thanks,
Juan

cgokmen, 2022-03-16:
@alanlou perhaps you can help here.

mjlbach, 2022-03-16:
It will take some (minor) modification, but here is how you can do 3d reconstruction from the 3d sensor:

```python
import open3d
import open3d as o3d
import open3d.core as o3c
import numpy as np

# Fixing the frustum clearing issue

# 
# We don't have this functionality supported, but since you are using the python API I guess you can do this manually:
#
# * you first create a frustum volume in the camera coordinate system by unprojecting the imaging plane to a list of discretized depths and reshape it to (N, 3) (only once)
# *then you rigid transform the volume to the world coordinate system (once per iteration)
# * finally you turn it into a Open3D Tensor and feed it to the function in need.

class Slam:
    def __init__(
            self,
            intrinsic,
            features = ["color", "seg", "ins_seg"],
            feature_sizes = [(3), (1), (1)],
            device="CUDA:0"):
        self.device = o3c.Device(device)
        voxel_size = 3.0 / 512

        self.features = features
        self.feature_sizes = feature_sizes

        attr_names = ["tsdf", "weight"] + features
        attr_sizes = [(1), (1) ] + feature_sizes
        attr_dtypes = [o3c.float32] * len(attr_sizes)

        self.vbg = o3d.t.geometry.VoxelBlockGrid(
            attr_names=attr_names,
            attr_dtypes=attr_dtypes,
            attr_channels=attr_sizes,
            voxel_size=voxel_size,
            block_resolution=16,
            block_count=50000,
            device=self.device,
        )
        self.trunc = voxel_size * 4
        self.depth_max = 5.0
        self.depth_scale = 1.0
        self.intrinsic = o3c.Tensor(intrinsic, o3c.Dtype.Float64)
        self.intrinsic_dev = o3c.Tensor(
            intrinsic, device=self.device, dtype=o3c.float32
        )

        pass

    @staticmethod
    def convert_opengl_to_world(ex):
        return np.array(
            [
                [ex[0, 0], ex[0, 1], ex[0, 2], ex[0, 3]],
                [-ex[1, 0], -ex[1, 1], -ex[1, 2], -ex[1, 3]],
                [-ex[2, 0], -ex[2, 1], -ex[2, 2], -ex[2, 3]],
                [0, 0, 0, 1],
            ]
        )

    def integrate(self, extrinsic, depth, features):
        extrinsic = self.convert_opengl_to_world(extrinsic)
        extrinsic = o3c.Tensor(extrinsic, dtype=o3c.float64)
        depth = open3d.t.geometry.Image(np.ascontiguousarray(depth))
        depth = depth.to(self.device)

[... 183 lines cut ...]
    o3c.cuda.synchronize()
    voxel_coords, voxel_indices = vbg.voxel_coordinates_and_flattened_indices(
        buf_indices
    )
    color = vbg.attribute("color").reshape((-1, 3))[voxel_indices]

    # Using active buf indices
    # color.shape = SizeVector[54960128, 3]
    # voxel_coords.shape = SizeVector[54960128, 3]

    # Using pcd to get voxel_coords
    # color.shape = SizeVector[48689152, 3]
    # voxel_coords.shape = SizeVector[48689152, 3]

    # Using pcd
    # pcd.point["colors"].shape = SizeVector[1510288, 3]

    # pcd = o3d.t.geometry.PointCloud()
    # pcd.point["positions"] = voxel_coords
    # pcd.point["colors"] = color / 255

    o3d.visualization.draw(pcd)

    # o3d.visualization.draw(vbg.extract_point_cloud())
```

WangJuan6, 2022-03-22:
Dear iGibson Team,
Thanks a lot, the code works.
I need to project the global semantic point cloud to the current frame, so I have to know how the coordinates are transformed. 
The action of the robot is set as forward or rotate, the extrinsic matrix can get from iGibson environment or calculate using orientation and position.
Can you help me to solve this problem?
My code is as follows:

```
import logging
import os
from sys import platform
import yaml
import matplotlib.pyplot as plt
import numpy as np

import igibson
from igibson.envs.igibson_env import iGibsonEnv
from igibson.render.profiler import Profiler

np.random.seed(2022)

def get_point_cloud_from_depth(Y_t, camera_matrix):
    xc = camera_matrix[0, 2]
    zc = camera_matrix[1, 2]
    fx = camera_matrix[0, 0]
    fy = camera_matrix[1, 1]
    scale = 1
    Y_t = Y_t.squeeze()
    grid_x, grid_z = np.meshgrid(np.arange(Y_t.shape[-1]),
                                 np.arange(Y_t.shape[-2] - 1, -1, -1))
    X_t = (grid_x[::scale, ::scale] - xc) * Y_t[::scale, ::scale] / fx
    Z_t = (grid_z[::scale, ::scale] - zc) * Y_t[::scale, ::scale] / fy
    XYZ = np.concatenate((X_t[..., np.newaxis], Y_t[::scale, ::scale][..., np.newaxis]
                          , Z_t[..., np.newaxis]), axis=2)
    return XYZ

def main():
    # config_filename = os.path.join(igibson.example_config_path, "husky_nav.yaml")
    config_filename = os.path.join(igibson.example_config_path, "turtlebot_nav.yaml")
    config_data = yaml.load(open(config_filename, "r"), Loader=yaml.FullLoader)
    config_data["vertical_fov"] = 90
    config_data["output"] = ["rgb", "depth"]
    # Reduce texture scale for Mac.
    if platform == "darwin":
        config_data["texture_scale"] = 0.5
    env = iGibsonEnv(config_file=config_data, mode="headless")

    env.simulator.robots[0].set_position([10,10,0])
    camera_matrix = env.simulator.renderer.get_intrinsics()

    for j in range(100):
        logging.info("Resetting environment")
        env.reset()
        for i in range(300):
            with Profiler("Environment action step"):
                # action = env.action_space.sample()
                # state, reward, done, info = env.step([0, 3, 1, 6])
                state, reward, done, info = env.step([1, 1])
                depth = state["depth"]
                extrinsics = env.simulator.renderer.V

                # depth to pcd
                point_cloud = get_point_cloud_from_depth(depth, camera_matrix)

                # transformation using extrinsics
                R_m = extrinsics[:3, :3]
                T_m = extrinsics[:3, 3:]
                point_cloud[..., 0] -= T_m[0]
                point_cloud[..., 1] -= T_m[1]
                point_cloud[..., 2] -= T_m[2]
                point_cloud_world_coordinate = R_m.T @ point_cloud.reshape(-1,3).transpose(1,0)
                point_cloud_world_coordinate = point_cloud_world_coordinate.transpose(1,0)

                # transformatin using position and orientation
                # current_rotation = env.simulator.robots[0].eyes.get_orientation()
                # current_position = env.simulator.robots[0].eyes.get_position()
                # from scipy.spatial.transform import Rotation as R
                # r = R.from_quat(current_rotation)
                # rotation_matrix = r.as_matrix()
                # point_cloud[..., 0] -= current_position[0]
                # point_cloud[..., 1] -= current_position[1]
                # point_cloud[..., 2] -= current_position[2]
                # point_cloud_world_coordinate = rotation_matrix.T @ point_cloud.reshape(-1, 3).transpose(1, 0)
                # point_cloud_world_coordinate = point_cloud_world_coordinate.transpose(1, 0)

                # combine
                if i==0:
                    point_cloud_global = point_cloud_world_coordinate
                else:
                    point_cloud_global = (np.append(point_cloud_global, point_cloud_world_coordinate, axis=0))

                # visualization
                ax = plt.figure(0).add_subplot(projection='3d')
                try:
                    ax.plot(np.ndarray.flatten(point_cloud[::, ::, 0]), np.ndarray.flatten(point_cloud[::, ::, 1]),
                            np.ndarray.flatten(point_cloud[::, ::, 2]), 'b.', markersize=0.2)
                except:
                    ax.plot(np.ndarray.flatten(point_cloud[::, 0]), np.ndarray.flatten(point_cloud[::, 1]),
                            np.ndarray.flatten(point_cloud[::, 2]), 'b.', markersize=0.2)
                plt.title('point cloud')

                ax = plt.figure(1).add_subplot(projection='3d')
                try:
                    ax.plot(np.ndarray.flatten(point_cloud_global[::, ::, 0]),
                            np.ndarray.flatten(point_cloud_global[::, ::, 1]),
                            np.ndarray.flatten(point_cloud_global[::, ::, 2]), 'b.', markersize=0.2)
                except:
                    ax.plot(np.ndarray.flatten(point_cloud_global[::, 0]),
                            np.ndarray.flatten(point_cloud_global[::, 1]),
                            np.ndarray.flatten(point_cloud_global[::, 2]), 'b.', markersize=0.2)
                plt.title('point_cloud_global')

                plt.show()

if __name__ == '__main__':
    main()
```

thanks,
Juan