StanfordVL/iGibson #179: Could you please update your tutorial for ros integration?

MRWANG995, 2022-03-17:
the demo is using ros1, turtlebot1 and python 2.7,  which are all out of date.
By using miniconda env based on python2.7, you even can not properly install igibson2!!

cgokmen, 2022-03-17:
iGibson in its current form is not compatible with Python 2.7. We are looking into re-enabling backwards compatibility, but are you able to switch to a Python 3 version of ROS? Would updating the tutorial to work with that be helpful for you?

MRWANG995, 2022-03-20:
> 

that would be of great favor for me.

vbalaji21, 2022-05-05:
@cgokmen and @MRWANG995 We have integrated ROS noetic (Python 3) with igibson 2. We are able to run the robot with move_base ROS navigation stack (Turtlebot and Fetch). Let us know if you need any help. We will happy to share our learnings with you :) . It may also help you to update the tutorial. Let us know :)

roberto-martinmartin, 2022-05-07:
Hi @vbalaji21 , please, try the new version we just released and tell us whether it presents any problem with you. Thanks!

cgokmen, 2022-05-07:
The release has not gone through yet - we'll push today. Will update here.

vbalaji21, 2022-05-08:
@roberto-martinmartin and @cgokmen . Thanks a lot for letting me know. Please update me once it is pushed, I will try it out.

This week, I had some problems with loading Pedestrians. I made some fixes and it is working now. I am not sure whether that is also taken into account in the new release. If not, I can share you the fixes I did. My aim (along with Prof. Rachid and Phanu Teja in LAAS, CNRS) is to study human aware social navigation and later extend with learning capabilities. 

My next 2 steps are -
1) Check VR immersion of human and arm (head data) to conduct user studies with our planner
2) Import a UAV into iGibson 

We are considering iGibson as our potential future platform for most of our development. We are hoping to do the above tasks as soon as possible and continue our development. We would be very happy to get any information or help on these 2 tasks.

cgokmen, 2022-05-08:
We definitely have well-maintained and functional VR capability right now (this is the feature we ourselves have used the most). You can try the VR demo in https://github.com/StanfordVL/iGibson/blob/master/igibson/examples/vr/vr_simple_demo.py using a SteamVR-installed computer (this demo was tested using a HTC Vive headset and Valve Index controller but I believe we support most devices).

As for the UAV we also have https://github.com/StanfordVL/iGibson/blob/master/igibson/robots/legacy/quadrotor_robot.py but this is a not-so-realistic quadrotor that used to be functional but is currently not maintained (e.g. it will probably take some code editing to make it work with the new Robot interface). Should be simple enough - the above file should end up looking like this one: https://github.com/StanfordVL/iGibson/blob/master/igibson/robots/ant.py

cgokmen, 2022-05-08:
Oh, also https://github.com/utiasDSL/gym-pybullet-drones has some interesting pybullet-based UAV simulation stuff (dynamics, control) that perhaps you could try using on top of iGibson (most of the code should be compatible and not too hard to hook up)

cgokmen, 2022-05-09:
We have just cut the release for iGibson 2.2.0 - please let us know if you need anything else, and feel free to reopen this thread if you have any further questions.

closed 2022-05-09 by cgokmen

vbalaji21, 2022-05-09:
Thanks a lot for the update and suggestions regarding UAV and VR.

I have 2 questions -

1) How to make any robot visible in REGISTERED_ROBOTS? I haave attached a screenshot of this error. Please let me know.
[image]

I tried importing UAV hector_quadcopter https://github.com/tu-darmstadt-ros-pkg/hector_quadrotor yesterday from here.
It is on ROS kinetic but tried porting it for ROS noetic.
I found that the controller classes for UAV needs to be ported.
I did not find controller classes for UAV in iGibson 1.0 as well.
I saw that you are using locomotion robot class and basecontroller for UAV if I am correct.

2) Does your comment on the realistic/functionality of Quadrotor mean that it doesn't work properly in iGibson 1.0 or the controller is primitive to make the Quadrotor work realistically ?

Thanks a lot for your comment about the Quadrorot/UAV suggestion. It will save a lot of time for us in the decision :)

vbalaji21, 2022-05-12:
> Hi @vbalaji21 , please, try the new version we just released and tell us whether it presents any problem with you. Thanks!

@roberto-martinmartin and @cgokmen , I tried the new version by installing iGibson in the compiling from source method but it failed to build in pip install -e .   com
[image]
[image]

Let me know

mand.
I have attached the error log screenshot below.

roberto-martinmartin, 2022-05-12:
could it be that you didn't `git clone` recursively?

vbalaji21, 2022-05-16:
@roberto-martinmartin Thank you the problem was indeed I didn't do git clone recursively.

I have another doubt, I followed the instructions but I did not understand about the Python Path step as it has 4 paths and I don't know in which file I need to update them. Can you please let me know ?

I think I am facing this problem with the path (Python path).
I have attached the screenshot below.

[image]