StanfordVL/iGibson #180: Trained model only making the turtlebot spinning around

genkv, 2022-03-18:
Hi, I have another question about the stable baselines example :)
I ran the stable_baselines3_example.py to generate a cktp model. And I want to see how it performs on the robot. Then I found out the trained model is only making the turtle bot spinning around and not actually trying to reach the target. Even though the trained model has a significantly higher mean reward -0.22 +/ 0.69 compared to before training (-10)

Here's the code I use to visualize the trained model. Is there something wrong with my code or making the robot spinning is the expected behavior?

```
config_file = "turtlebot_nav.yaml"
env = iGibsonEnv(
        config_file=os.path.join(igibson.configs_path, config_file),
        mode="gui_interactive",
        action_timestep=1 / 10.0,
        physics_timestep=1 / 120.0
)

s = env.simulator
s.viewer.initial_pos = [-2, 1.4, 1.2]
s.viewer.initial_view_direction = [0.6, -0.8, 0.1]
s.viewer.reset_viewer()

model = PPO.load(sys.argv[1], env=env)
obs = env.reset()

for _ in range(1000):
        with Profiler("Simulator step"):
            action, _states = model.predict(obs, deterministic=True)
            obs, rewards, dones, info = env.step(action)
            rgb = s.renderer.render_robot_cameras(modes=("rgb"))
```

thank you

genkv, 2022-03-21:
update:

One thing I noticed is that before any training, the evaluation will produce a -10 mean reward if you just let the robot sample random actions. After 1 million steps of training, the mean reward is improved to around 0. The reason being the PPO mode  discovered that only spinning around is the safest policy because it won't collide with anything (zero penalties), but at the same time it also won't get any positive rewards (zero rewards). That's why the mean reward post-training is around zero.

Can anyone from the team confirm if my analysis is correct? Then is there any example to demonstrate the training of a meaningful model that actually make use the observations to complete the PointNav task?

Thank you

mjlbach, 2022-03-21:
FWIW I am regularly training visual object navigation in iGibson, I'm not sure if I can share the code since it's for an active project, but what I've found is useful for navigation tasks are:

* Ensure all doors are opened (or removed)
* Use a geodesic (potential) reward with a magnitude ~ half that of the discovery reward
* Use an RGBD input space
* Use a discrete action space (forward, turn left, turn right is sufficient) where turn left/turn right halt forward motion
* Train with 8 environments for approximately 100,000 environment steps (I use ray).

FWIW I am able to train more advanced tasks, like pick/place, with this setup:

https://user-images.githubusercontent.com/13316262/159354926-129470f2-79c7-4edb-8407-284ce916a372.mp4

genkv, 2022-03-21:
@mjlbach Thank you for the detailed feedback!

I think the environment setup in stable_baselines3_example.py (which uses turtlebot_nav.yaml) meets the first 3 criteria you mentioned. It does, however, use a continuous action space on the Differential drive controller of the turtlebot:

```
robot:
  name: Turtlebot
  action_type: continuous
  action_normalize: true
  base_name: null
  scale: 1.0
  self_collision: false
  rendering_params: null
  controller_config:
    base:
      name: DifferentialDriveController
```

So maybe one thing I can work on is to modify this example to use discrete action space.

About the last suggestion. I added more RAM to make up 64 gb in total. But the process would still get killed by daemon when I train with 8 environments for 1 million steps (default in stable_baselines3_example.py). My devbox only has so many slots that I cannot add more RAM. So I trained the model with only 2 environments for 1 million steps. Maybe the number of environments also contributes to the result of "only-spinning-model"? I thought it would only affect efficiency and training time.
In your comment you mentioned 100k instead of 1 million. I will give it a try with 8 environments.

Lastly, can you confirm if there's no major flaw with the code I posted when it comes to visualizing a trained model?

Thank you so much.

genkv, 2022-03-23:
Update: the stable baselines from example will eat up RAMs after the 1st iteration. I can confirm that this .py example cannot be ran out of box. Some modifications need to be made. I'm new to RL so still slowing learning.

Lowering the number of environments is the quickest way, but will compromise the quality of the model, and end up letting the trained model only spinning around like described.

mjlbach, 2022-04-13:
Here's an example repo for training an object navigation task:
https://github.com/stanfordVL/ig_navigation

Results (after about an hour of training on 8 environments)
[image]

https://user-images.githubusercontent.com/13316262/163080088-eff0aeec-7
bc1-4503-a67f-892c1dbce43f.mp4

https://user-images.githubusercontent.com/13316262/163080111-be07a3d0-efc3-4236-ac2d-be249ec05824.mp4

https://user-images.githubusercontent.com/13316262/163080154-a4e63255-e6f2-44ab-b425-05c22aef459a.mp4

There's a small bug in the potential reward computation (pathfinding breaks through doorways) that about halves training efficiency, but this should get you all started.

closed 2022-04-13 by mjlbach