StanfordVL/iGibson #183: How to Initialize the starting position of robot in motion planning demo

vbalaji21, 2022-03-24:
Hi,

I am using iGibson 2.0 and running the motion_planning demo. Sometimes the robot starts inside a small room. It works well when the robot is a good room with space. please let me know how do I give initial starting position of the robot ?

ChengshuLi, 2022-03-28:
Hi, are you running this example?
https://github.com/StanfordVL/iGibson/blob/master/igibson/examples/robots/motion_planning_example.py

It seems that the robot initial position is always [0, 0, 0]? 
https://github.com/StanfordVL/iGibson/blob/b1521122c1d39dbf392012b6fc0af99cbc62fcb7/igibson/examples/robots/motion_planning_example.py#L97

In any case, you can set the position by doing 
```
x, y, z = # any desired position you want
env.land(env.robots[0], [x, y, z], [0, 0, 0])
```

vbalaji21, 2022-04-06:
Hi, Yes. I am running the same example.
But everytime I start the motion planning example, the robot has a new position and most of  the time the robot initial position is struck in obstacles and I have to start the program again multiple times for the robot to have a good free initial postion without obstacles.

Is there any other parameter in the code interfering with the starting position of the robot ?

ChengshuLi, 2022-06-10:
I am not exactly sure I understand the issue here. Did you use `programmatic=True` or `programmatic=False`?

chenguanqi, 2023-07-31:
After using the env.land(robot, pos, orn) function, you should immediately use the robot.tuck() function. Then the robot will be reset to the initial position you defined.
```python
env.land(env.robots[0], pos=[0, 0, 0], orn=[0, 0, 0])
env.robots[0].tuck()
```