StanfordVL/iGibson #184: Object texture not loading

SophieHsu, 2022-03-24:
[image]

Hi,

We are experiencing a similar issue as #116. As seen in the image, the objects we load from the ig_dataset does not include its texture. I have ran the [load_object.py](https://github.com/StanfordVL/iGibson/blob/master/igibson/examples/objects/load_objects.py) example and the texture of the objects did show up in the "gui_interactive" mode. However, after comparing my code with the [load_object.py](https://github.com/StanfordVL/iGibson/blob/master/igibson/examples/objects/load_objects.py) example, I still fail to find what might be missing on my end that stops me from loading the object's texture. Here is the code we are running that results in this issue:
```python
import argparse
import os
import igibson
from igibson.envs.igibson_env import iGibsonEnv
from igibson.objects.articulated_object import URDFObject
    
def run_example(args):
    nav_env = iGibsonEnv(
        config_file=args.config, mode="gui_interactive", action_timestep=1.0 / 12, physics_timestep=1.0 / 12, use_pb_gui=True
    ) 
    robot_x, robot_y = 2, 5
    nav_env.land(nav_env.robots[0], [robot_x-4.5, robot_y-4.5, 0], [0, 0, 0])
    
    filepath = os.path.join(igibson.ig_dataset_path, "objects/bottom_cabinet/46452/46452.urdf")
    obj = URDFObject(filepath, model_path="/".join(filepath.split("/")[:-1]))
    nav_env.simulator.import_object(obj)
    nav_env.land(obj, (0.599, -0.5, 0), (0, 0, -1.5707))

    filepath = os.path.join(igibson.ig_dataset_path, "objects/fridge/11712/11712.urdf")
    obj = URDFObject(filepath, model_path="/".join(filepath.split("/")[:-1]))
    nav_env.simulator.import_object(obj)
    nav_env.land(obj, (0.599, -2.5, 0), (0, 0, -1.5707))

    filepath = os.path.join(igibson.ig_dataset_path, "objects/sink/kitchen_sink/kitchen_sink.urdf")
    obj = URDFObject(filepath, model_path="/".join(filepath.split("/")[:-1]))
    nav_env.simulator.import_object(obj)
    nav_env.land(obj, (0.599, 1.5, 0), (0, 0, -1.5707))

    try:
        while True:
            continue
    except:
        pass

if __name__ == "__main__":
    parser = argparse.ArgumentParser()
    parser.add_argument(
        "--config",
        "-c",
        default="../fetch_motion_planning_test.yaml",
        help="which config file to use [default: use yaml files in examples/configs]",
    )

    args = parser.parse_args()
    run_example(args)
```
The config file we use is the one below:
```config
scene: empty
scene_id: Beechwood_1_int
build_graph: true
load_texture: true
pybullet_load_texture: true
trav_map_type: no_obj
trav_map_resolution: 0.1
trav_map_erosion: 3
should_open_all_doors: true

# domain randomization
texture_randomization_freq: null
object_randomization_freq: null

# robot
robot:
  name: Fetch
  action_type: continuous
  action_normalize: true
  base_name: null
  scale: 1.0
  self_collision: true
  rendering_params: null
  rigid_trunk: false
  default_trunk_offset: 0.365
  default_arm_pose: vertical
  controller_config:
    base:
      name: DifferentialDriveController
    arm:
      name: InverseKinematicsController
      kv: 2.0
    gripper:
      name: ParallelJawGripperController
      mode: binary
    camera:
      name: JointController
      use_delta_commands: False

# task 
#task: dummy_lsi_3d
load_object_categories: [
]

# reward
potential_reward_weight: 1.0
prismatic_joint_reward_scale: 3.0
revolute_joint_reward_scale: 1.0

# discount factor
discount_factor: 0.99

# termination condition
max_step: 500
max_collisions_allowed: 500

# misc config
initial_pos_z_offset: 0.1
collision_ignore_link_a_ids: [0, 1, 2]  # ignore collisions with these robot links

# sensor spec
output: [rgb, depth, scan, occupancy_grid]
# image
# Primesense Carmine 1.09 short-range RGBD sensor
# http://xtionprolive.com/primesense-carmine-1.09
fisheye: false
image_width: 512
image_height: 512
vertical_fov: 90
# depth
depth_low: 0.35
depth_high: 3.0
# scan
# SICK TIM571 scanning range finder
# https://docs.fetchrobotics.com/robot_hardware.html
# n_horizontal_rays is originally 661, sub-sampled 1/3
n_horizontal_rays: 220
n_vertical_beams: 1
laser_linear_range: 25.0
laser_angular_range: 220.0
min_laser_dist: 0.05
laser_link_name: laser_link

# sensor noise
depth_noise_rate: 0.0
scan_noise_rate: 0.0

# visual objects
visible_target: true
visible_path: false
```
I am using an Ubuntu 20.04. The python version is 3.8.12. The igibson version is 2.1.0. I am running with conda. Here is my environment file:
```
name: igibson
channels:
  - pytorch
  - defaults
dependencies:
  - _libgcc_mutex=0.1
  - _openmp_mutex=4.5
  - _pytorch_select=0.1
  - blas=1.0
  - bottleneck=1.3.2
  - brotli=1.0.9
  - bzip2=1.0.8
  - ca-certificates=2022.3.18
  - certifi=2021.10.8
  - cffi=1.14.6
  - cpuonly=2.0
  - cudatoolkit=10.1.243
  - cycler=0.11.0
  - dbus=1.13.18
  - expat=2.4.1
  - ffmpeg=4.3
  - fontconfig=2.13.1
  - fonttools=4.25.0
  - freetype=2.11.0
  - giflib=5.2.1
  - glib=2.69.1
  - gmp=6.2.1
  - gnutls=3.6.15
  - gst-plugins-base=1.14.0
  - gstreamer=1.14.0
  - icu=58.2
  - intel-openmp=2019.4
  - jpeg=9d
  - kiwisolver=1.3.1
  - lame=3.100
  - lcms2=2.12
  - ld_impl_linux-64=2.35.1
  - libffi=3.3
  - libgcc-ng=9.3.0
  - libgfortran-ng=7.5.0
  - libgfortran4=7.5.0
  - libgomp=9.3.0
  - libiconv=1.15
  - libidn2=2.3.2
  - libmklml=2019.0.5
  - libpng=1.6.37
  - libstdcxx-ng=9.3.0
  - libtasn1=4.16.0
  - libtiff=4.2.0
  - libunistring=0.9.10
  - libuuid=1.0.3
  - libuv=1.40.0
  - libwebp=1.2.0
  - libwebp-base=1.2.0
  - libxcb=1.14
  - libxml2=2.9.12
  - lz4-c=1.9.3
  - matplotlib=3.5.0
  - matplotlib-base=3.5.0
  - mkl=2020.2
  - mkl-service=2.3.0
  - mkl_fft=1.3.0
  - mkl_random=1.1.1
  - munkres=1.1.4
  - ncurses=6.3
  - nettle=3.7.3
  - ninja=1.10.2
  - numexpr=2.7.3
  - numpy=1.19.2
  - numpy-base=1.19.2
  - olefile=0.46
  - openh264=2.1.1
  - openssl=1.1.1l
  - packaging=21.3
  - pandas=1.3.4
  - pcre=8.45
  - pip=21.2.4
  - pycparser=2.21
  - pyqt=5.9.2
  - python=3.8.12
  - python-dateutil=2.8.2
  - pytorch-mutex=1.0
  - qt=5.9.7
  - readline=8.1
  - scipy=1.6.2
  - seaborn=0.11.2
  - setuptools=58.0.4
  - sip=4.19.13
  - six=1.16.0
  - sqlite=3.36.0
  - tk=8.6.11
  - tornado=6.1
  - typing-extensions=3.10.0.2
  - typing_extensions=3.10.0.2
  - wheel=0.37.0
  - xz=5.2.5
  - zlib=1.2.11
  - zstd=1.4.9
  - pip:
    - bddl==1.0.1
    - future==0.18.2
    - iniconfig==1.1.1
    - jupytext==1.13.6
    - markdown-it-py==1.1.0
    - mdit-py-plugins==0.3.0
    - opencv-python==4.1.2.30
    - pexpect==4.8.0
    - pillow==8.4.0
    - pluggy==1.0.0
    - progressbar==2.5
    - py==1.11.0
    - pybullet-svl==3.1.6.4
    - pyparsing==3.0.6
    - pytest==6.2.5
    - pytz==2021.3
    - pyyaml==6.0
    - toml==0.10.2
    - torch==1.9.0+cu111
    - torchaudio==0.9.0
    - torchvision==0.10.0+cu111
    - urllib3==1.26.8
```
Thank you!

mjlbach, 2022-03-24:
The ig_dataset is encrypted, you won't be able to view the texture in the pybullet viewer.

SophieHsu, 2022-03-24:
Hi @mjlbach, 

Thank you for getting back to me so quickly. I would then like to ask if I can actually view the texture in the external viewers instead. Since that in the [load_objects.py](https://github.com/StanfordVL/iGibson/blob/master/igibson/examples/objects/load_objects.py) example, I was able to see the texture of the objects in the external viewer window (as shown in the image below).
[image]

But it seems to not show texture in the external windows when I execute my code (the python file included in the first post).
[image]

Is there something I am missing?
Thank you!

roberto-martinmartin, 2022-03-25:
Sorry, I thought that would answer your question but I see it is different. Feel free to reopen the issue.

Could you do two things?
1. activate debug level of logging in your application (add `logging.basicConfig(level=logging.DEBUG` at the beginning of your "main")
2. copy here the terminal output

My current hints: there may be some arguments you need to set in the URDFObject constructions and/or it is an issue when loading external objects that are not encrypted

Thanks

roberto-martinmartin, 2022-03-25:
Also, can you move the viewer's point of view to see if the texture is missing or just the entire objects are not loaded? thanks!

SophieHsu, 2022-03-25:
@roberto-martinmartin No worries! Thank you for reopening the issue :)
I have moved and zoomed in in the viewer's point of view and it seems like the objects are not loaded. Here's the screenshot of the external viewer window:
[image]

And here is the debug level of logging:
```
INFO:igibson.render.mesh_renderer.get_available_devices:Device 0 is available for rendering
INFO:igibson.render.mesh_renderer.get_available_devices:Command '['/project/lsi_3d/iGibson/igibson/render/mesh_renderer/build/test_device', '1']' returned non-zero exit status 1.
INFO:igibson.render.mesh_renderer.get_available_devices:Device 1 is not available for rendering
startThreads creating 1 threads.
starting thread 0
started thread 0 
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
X11 functions dynamically loaded using dlopen/dlsym OK!
Creating context
Created GL 3.3 context
Direct GLX rendering context obtained
Making context current
GL_VENDOR=NVIDIA Corporation
GL_RENDERER=NVIDIA GeForce RTX 3090/PCIe/SSE2
GL_VERSION=3.3.0 NVIDIA 470.103.01
GL_SHADING_LANGUAGE_VERSION=3.30 NVIDIA via Cg compiler
pthread_getconcurrency()=0
Version = 3.3.0 NVIDIA 470.103.01
Vendor = NVIDIA Corporation
Renderer = NVIDIA GeForce RTX 3090/PCIe/SSE2
b3Printf: Selected demo: Physics Server
startThreads creating 1 threads.
starting thread 0
started thread 0 
MotionThreadFunc thread started
ven = NVIDIA Corporation
ven = NVIDIA Corporation
INFO:igibson.scenes.scene_base:Loading scene...
INFO:igibson.scenes.scene_base:Scene loaded!
/home/jasonfeng/.local/lib/python3.8/site-packages/gym/spaces/box.py:73: UserWarning: WARN: Box bound precision lowered by casting to float32
  logger.warn(
DEBUG:trimesh:loaded <trimesh.Trimesh(vertices.shape=(64, 3), faces.shape=(108, 3))> using load_obj
DEBUG:trimesh:load_mesh executed in 0.0029 seconds.
DEBUG:trimesh:loaded <trimesh.Trimesh(vertices.shape=(165, 3), faces.shape=(300, 3))> using load_obj
DEBUG:trimesh:load_mesh executed in 0.0009 seconds.
DEBUG:trimesh:loaded <trimesh.Trimesh(vertices.shape=(98, 3), faces.shape=(164, 3))> using load_obj
DEBUG:trimesh:load_mesh executed in 0.0007 seconds.
ERROR:igibson.render.mesh_renderer.mesh_renderer_cpu:Using optimized renderer and optimization process is already excuted, cannot add new objects
ERROR:igibson.render.mesh_renderer.mesh_renderer_cpu:Using optimized renderer and optimization process is already excuted, cannot add new objects
ERROR:igibson.render.mesh_renderer.mesh_renderer_cpu:Using optimized renderer and optimization process is already excuted, cannot add new objects
ERROR:igibson.render.mesh_renderer.mesh_renderer_cpu:Using optimized renderer and optimization process is already excuted, cannot add new objects
DEBUG:trimesh:loaded <trimesh.Trimesh(vertices.shape=(95, 3), faces.shape=(166, 3))> using load_obj
DEBUG:trimesh:load_mesh executed in 0.0010 seconds.
DEBUG:trimesh:loaded <trimesh.Trimesh(vertices.shape=(109, 3), faces.shape=(192, 3))> using load_obj
DEBUG:trimesh:load_mesh executed in 0.0008 seconds.
DEBUG:trimesh:loaded <trimesh.Trimesh(vertices.shape=(256, 3), faces.shape=(472, 3))> using load_obj
DEBUG:trimesh:load_mesh executed in 0.0012 seconds.
ERROR:igibson.render.mesh_renderer.mesh_renderer_cpu:Using optimized renderer and optimization process is already excuted, cannot add new objects
ERROR:igibson.render.mesh_renderer.mesh_renderer_cpu:Using optimized renderer and optimization process is already excuted, cannot add new objects
ERROR:igibson.render.mesh_renderer.mesh_renderer_cpu:Using optimized renderer and optimization process is already excuted, cannot add new objects
ERROR:igibson.render.mesh_renderer.mesh_renderer_cpu:Using optimized renderer and optimization process is already excuted, cannot add new objects
DEBUG:trimesh:loaded <trimesh.Trimesh(vertices.shape=(10685, 3), faces.shape=(18654, 3))> using load_obj
DEBUG:trimesh:load_mesh executed in 0.0218 seconds.
ERROR:igibson.render.mesh_renderer.mesh_renderer_cpu:Using optimized renderer and optimization process is already excuted, cannot add new objects
ERROR:igibson.render.mesh_renderer.mesh_renderer_cpu:Using optimized renderer and optimization process is already excuted, cannot add new objects
ERROR:igibson.render.mesh_renderer.mesh_renderer_cpu:Using optimized renderer and optimization process is already excuted, cannot add new objects
ERROR:igibson.render.mesh_renderer.mesh_renderer_cpu:Using optimized renderer and optimization process is already excuted, cannot add new objects
ERROR:igibson.render.mesh_renderer.mesh_renderer_cpu:Using optimized renderer and optimization process is already excuted, cannot add new objects

```

roberto-martinmartin, 2022-03-25:
ok, from the logs it looks like the renderer has been stepped already when you try to load the objects. Because the first time we render, we optimize, you can't add more objects later. One simple option is to not use the optimized renderer but that is not ideal as you will lose performance.

The question is where in your code the renderer is called. It may be the `land` command that internally steps the simulator for a couple of steps. Could you try first loading all the objects and move the landing calls to execute after all have been loaded?

roberto-martinmartin, 2022-03-25:
btw, I'm curious, were the 'ERROR' messages being printed in the terminal before we set the level to debug?

SophieHsu, 2022-03-25:
Hi @roberto-martinmartin,

It is now working! Thank you! It was the `land` command that internally steps the simulator. Once I moved every `land` command after every object was imported, the objects, along with their texture, showed up in the external viewer.
[image]
```
def run_example(args):
    nav_env = iGibsonEnv(
        config_file=args.config, mode="gui_interactive", action_timestep=1.0 / 12, physics_timestep=1.0 / 12, use_pb_gui=True
    ) 
    robot_x, robot_y = 2, 5
    
    filepath = os.path.join(igibson.ig_dataset_path, "objects/bottom_cabinet/46452/46452.urdf")
    obj = URDFObject(filepath, model_path="/".join(filepath.split("/")[:-1]))
    nav_env.simulator.import_object(obj)

    filepath = os.path.join(igibson.ig_dataset_path, "objects/fridge/11712/11712.urdf")
    obj2 = URDFObject(filepath, model_path="/".join(filepath.split("/")[:-1]))
    nav_env.simulator.import_object(obj2)

    filepath = os.path.join(igibson.ig_dataset_path, "objects/sink/kitchen_sink/kitchen_sink.urdf")
    obj3 = URDFObject(filepath, model_path="/".join(filepath.split("/")[:-1]))
    nav_env.simulator.import_object(obj3)

    try:
        nav_env.land(nav_env.robots[0], [robot_x-4.5, robot_y-4.5, 0], [0, 0, 0])
        nav_env.land(obj, (0.599, -0.5, 0), (0, 0, -1.5707))
        nav_env.land(obj2, (0.599, -2.5, 0), (0, 0, -1.5707))
        nav_env.land(obj3, (0.599, 1.5, 0), (0, 0, -1.5707))
        
        while True:
            nav_env.simulator.step()
    except:
        pass
```

And to answer the previous comment, the 'ERROR' did not exist before setting the level to debug. The following is the terminal output when I comment out the prior added logging command.
```
startThreads creating 1 threads.
starting thread 0
started thread 0 
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
X11 functions dynamically loaded using dlopen/dlsym OK!
Creating context
Created GL 3.3 context
Direct GLX rendering context obtained
Making context current
GL_VENDOR=NVIDIA Corporation
GL_RENDERER=NVIDIA GeForce RTX 3090/PCIe/SSE2
GL_VERSION=3.3.0 NVIDIA 470.103.01
GL_SHADING_LANGUAGE_VERSION=3.30 NVIDIA via Cg compiler
pthread_getconcurrency()=0
Version = 3.3.0 NVIDIA 470.103.01
Vendor = NVIDIA Corporation
Renderer = NVIDIA GeForce RTX 3090/PCIe/SSE2
b3Printf: Selected demo: Physics Server
startThreads creating 1 threads.
starting thread 0
started thread 0 
MotionThreadFunc thread started
ven = NVIDIA Corporation
ven = NVIDIA Corporation
/home/jasonfeng/.local/lib/python3.8/site-packages/gym/spaces/box.py:73: UserWarning: WARN: Box bound precision lowered by casting to float32
  logger.warn(
Using optimized renderer and optimization process is already excuted, cannot add new objects
Using optimized renderer and optimization process is already excuted, cannot add new objects
Using optimized renderer and optimization process is already excuted, cannot add new objects
Using optimized renderer and optimization process is already excuted, cannot add new objects
Using optimized renderer and optimization process is already excuted, cannot add new objects
Using optimized renderer and optimization process is already excuted, cannot add new objects
Using optimized renderer and optimization process is already excuted, cannot add new objects
Using optimized renderer and optimization process is already excuted, cannot add new objects
Using optimized renderer and optimization process is already excuted, cannot add new objects
Using optimized renderer and optimization process is already excuted, cannot add new objects
Using optimized renderer and optimization process is already excuted, cannot add new objects
Using optimized renderer and optimization process is already excuted, cannot add new objects
Using optimized renderer and optimization process is already excuted, cannot add new objects
```

Thank you for helping me resolve the issue :)

closed 2022-03-25 by SophieHsu