StanfordVL/iGibson #188: Current state of motion primitives

ju-kreber, 2022-03-28:
Hello, I'm trying to reconstruct results of the BEHAVIOR paper, in particular with reinforcement learning agents using discrete actions / motion primitives. I've come to notice that the iGibon implementation of `BehaviorMPEnv` (`igibson.envs.behavior_mp_env`) , which is used in the  `replay_demo_with_action_primitives_example.py` in the current version of Behavior, is no longer existent. The file has been removed in `4235eef`. I'm now wondering how to use motion primitives (without motion planning) in the current version. Is this no longer supported?
Even if it's not supported anymore, could you hint me at a commit I could use where the functionality is still there? I've already tried multiple commits, just before the file was removed, version 2.0.5 and version 2.0.3, but none of them ran successfully due to different errors. Sadly, I could not get it working.

I'd really appreciate your advise on this, thanks!

cgokmen, 2022-04-10:
Hello,

Thanks for reaching out - as you noticed we removed this motion primitive interface and are working on a new, cleaner and more effective one. Unfortunately we have no immediate date estimate for when this will be released although I would expect it to be done in May. The original interface used for those experiments was discontinued due to its incompatibility with many later refactors. I recommend trying out the earliest 2.0 releases (and will check again for you when I am at a computer later) but again this interface is not officially supported in any public version of iGibson.

Thanks,
Cem

ju-kreber, 2022-04-11:
Thank you for the information! I'm looking forward to the new implementation. :)

closed 2022-04-11 by ju-kreber

shirley-wu, 2023-10-14:
Hi, may I ask how is the current state for implementing BehaviorMPEnv with primitive actions? Thanks!