StanfordVL/iGibson #189: ERROR: Unable to initialize EGL

mukulkhanna, 2022-03-31:
Hi team, thank you for maintaining this project.

My iGibson installation went fine, but I am facing an issue that seems common among many iGibson beginners.

```
(igib) ➜  ~ python -m igibson.examples.environments.env_nonint_example

 _   _____  _  _
(_) / ____|(_)| |
 _ | |  __  _ | |__   ___   ___   _ __
| || | |_ || || '_ \ / __| / _ \ | '_ \
| || |__| || || |_) |\__ \| (_) || | | |
|_| \_____||_||_.__/ |___/ \___/ |_| |_|

********************************************************************************
Description:
    Creates an iGibson environment from a config file with a turtlebot in Rs (not interactive).
    It steps the environment 100 times with random actions sampled from the action space,
    using the Gym interface, resetting it 10 times.
    ********************************************************************************
INFO:igibson.render.mesh_renderer.get_available_devices:Command '['/home/mukul/iGibson/igibson/render/mesh_renderer/build/test_device', '0']' returned non-zero exit status 1.
INFO:igibson.render.mesh_renderer.get_available_devices:Device 0 is not available for rendering
WARNING:igibson.render.mesh_renderer.mesh_renderer_cpu:Device index is larger than number of devices, falling back to use 0
WARNING:igibson.render.mesh_renderer.mesh_renderer_cpu:If you have trouble using EGL, please visit our trouble shooting guideat http://svl.stanford.edu/igibson/docs/issues.html
libEGL warning: DRI2: failed to create dri screen
libEGL warning: DRI2: failed to create dri screen
ERROR: Unable to initialize EGL
```

I went through all the closed issues related to this, but nothing helped.
I also went through the [troubleshooting guide](http://svl.stanford.edu/gibson2/docs/issues.html) and things seemed to check out. Here are the outputs of some commands I ran to check the EGL installation:
- ```
  (igib) ➜  ~ ldconfig -p | grep EGL
	libEGL_nvidia.so.0 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libEGL_nvidia.so.0
	libEGL_mesa.so.0 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libEGL_mesa.so.0
	libEGL.so.1 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libEGL.so.1
	libEGL.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libEGL.so
  ```

- ```
  (igib) ➜  ~ nvidia-smi
  Thu Mar 31 15:28:55 2022       
  +-----------------------------------------------------------------------------+
  | NVIDIA-SMI 510.47.03    Driver Version: 510.47.03    CUDA Version: 11.6     |
  |-------------------------------+----------------------+----------------------+
  | GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
  | Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
  |                               |                      |               MIG M. |
  |===============================+======================+======================|
  |   0  NVIDIA GeForce ...  On   | 00000000:01:00.0  On |                  N/A |
  | 41%   39C    P8    20W / 215W |    331MiB /  8192MiB |      2%      Default |
  |                               |                      |                  N/A |
  +-------------------------------+----------------------+----------------------+
  ```
- Reinstalling after `USE_GLAD` set to `FALSE` didn't work either.

Please let me know if I can share any more information that could be helpful in debugging this. 

Thanks!

WJJLBJ, 2022-04-07:
hello, have you fixed the problem? could you tell me how to fix it? thank you!

CharlesAverill, 2022-04-07:
@WJJLBJ @mukulkhanna I usually get EGL issues when I reinstall an NVIDIA driver or forget to activate my conda env

mukulkhanna, 2022-04-07:
@CharlesAverill, thanks. I did recently reinstall my drivers, but my conda env is activated.

@WJJLBJ, I am still stuck with this issue.

mjlbach, 2022-04-07:
The EGL loading code is identical to https://github.com/erwincoumans/egl_example

What does `apt list --installed | grep libegl1` say?

WJJLBJ, 2022-04-07:
> 

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libegl1/bionic-updates,now 1.0.0-2ubuntu2.3 amd64 [installed]
libegl1-mesa/bionic-updates,now 20.0.8-0ubuntu1~18.04.1 amd64 [installed,auto-removable]

WJJLBJ, 2022-04-07:
@mjlbach 
WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libegl1/bionic-updates,now 1.0.0-2ubuntu2.3 amd64 [installed]
libegl1-mesa/bionic-updates,now 20.0.8-0ubuntu1~18.04.1 amd64 [installed,auto-removable]

mjlbach, 2022-04-07:
@WJJLBJ all of you could have slightly different issues so it might be your nvidia driver version. I would recommend pulling down and building the repo I linked and see if it works (it should not). Nvidia driver version can cause EGL to fail, missing libegl1 can cause it to fail, etc. You can try running our containers to narrow down the problem/see the minimum dependencies.

mjlbach, 2022-04-07:
Oh sorry, make sure you have `libegl-dev`

mukulkhanna, 2022-04-07:
Thanks @mjlbach for the context. I'm pasting my outputs to your suggested commands here to eliminate any obvious issues.

```
(igib) ➜  iGibson git:(master) ✗ apt list --installed | grep libegl1

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libegl1-mesa-dev/focal-security,focal-updates,now 21.2.6-0ubuntu0.1~20.04.2 amd64 [installed]
libegl1-mesa/focal-security,focal-updates,now 21.2.6-0ubuntu0.1~20.04.2 amd64 [installed]
libegl1/focal-security,focal-updates,now 1.3.2-1~ubuntu0.20.04.2 amd64 [installed,automatic]
(igib) ➜  iGibson git:(master) ✗ apt list --installed | grep libegl-dev

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libegl-dev/focal-security,focal-updates,now 1.3.2-1~ubuntu0.20.04.2 amd64 [installed]
```

mjlbach, 2022-04-07:
Hmm, then I don't know, I would try re-building egl example. The only times I was able to reproduce this were:

A. A bad series of nvidia drivers with a cgroups bug (460s)
B. Missing libegl-dev and it's dependencies

I'm using iGibson on Fedora 35 and Ubuntu 18.04/20.04 without issue, so not sure.

WJJLBJ, 2022-04-07:
@mjlbach 
thank you for you reply, i am using ubuntu18 and i dont know how to install libegl-dev since i have searched for it on Google, could you tell me how to install libegl-dev?

mukulkhanna, 2022-04-07:
> I would try re-building egl example.

Thanks @mjlbach. I'll try this and get back.

mjlbach, 2022-04-07:
Just `sudo apt install libegl-dev` should work?

WJJLBJ, 2022-04-07:
@mjlbach 
it does not work for me..
E: Unable to locate package libegl-dev
so i am wondering about it

mukulkhanna, 2022-04-07:
> `sudo apt install libegl-dev`
```
libegl-dev is already the newest version (1.3.2-1~ubuntu0.20.04.2).
```

mjlbach, 2022-04-07:
Perhaps 18.04 uses a different package name. I would recommend you try one of our containers if you just want to get started https://github.com/StanfordVL/iGibson/blob/master/docker/igibson/Dockerfile

mukulkhanna, 2022-04-12:
I was able to resolve this issue by downgrading my NVIDIA GPU driver from 510 to 470  😄 Therefore, closing this.

closed 2022-04-12 by mukulkhanna

JasonLiu324, 2022-07-30:
@WJJLBJ Hi, I encountered the same issue as you mentioned about the egl problem when using ubuntu 18.04. But then I found in Google that there is not ligegl-dev for ubuntu18.04, so I upgraded the system to 20.04. Then it finally works. And I also recommend you to install ros-noetic because ros would help you fix many egl dependency issues.

taylorlulu, 2024-02-29:
I faced the same problem. I was trying to run igibson on the server. The version of NVIDIA GPU driver is 535. The command `sudo apt install libegl-dev` and https://github.com/erwincoumans/egl_example  went just fine. 
[image]

[image]

But I still have the Error : Failed to EGL with glad.
[image]

I solved the problem by setting after USE_GLAD set to FALSE in https://github.com/StanfordVL/iGibson/blob/7d9598278b95ec1039e1f4366c66b45346521e44/igibson/render/CMakeLists.txt#L4 didn't work either.
The I reinstall by the command:
`python setup.py install`
That build the project again. The error disappeared but I still face the problem.